aa-proxy/aa-proxy-rs #1: Please provide proper installation guide

NabeelUppel, 2024-11-04:
Hi!
I'm keen to give this a shot, look quite interesting. I have used aawgd before but all of a sudden stopped working for me so I'm curious to see if this will work.

With that being said, I'm quite new to all of this and have no experience with Rust but could be a good opportunity to learn.

I see that you SSH into the pi to run the script but what OS is running on your pi to make it work?

manio, 2024-11-04:
Hi,
Just use [WirelessAndroidAutoDongle](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle) as a base and only replace aawgd with aa-proxy-rs.
Yesterday I added a Dockerfile and instruction how to built it. Can you try to build it first?

NabeelUppel, 2024-11-04:
I managed to build it with docker and have the binary but I'm unsure how to replace aawgd with it.

NabeelUppel, 2024-11-04:
I haven't managed to uninstall aawgd but I did drop the built aa-proxy-rs onto the same path as you had and I chmod +x the binary to get it to run

I then get Bluetooth error: Bluetooth operation not permitted: UUID already registered. 
I suspect it's because I haven't removed aawgd. 

I also noticed the binary is not persisted on SD card.

NabeelUppel, 2024-11-04:
Alright, I managed to remove aawgd from /usr/bin and also dropped aa-proxy-rs in the same folder

NabeelUppel, 2024-11-04:
[image]

Seems as tho the wifi disconnects as my PC switches back to my home wifi

NabeelUppel, 2024-11-04:
I'm assumming the failure is on line 107
[image]

NabeelUppel, 2024-11-04:
I apologize for so many comments but I thought this would be a good place to be verbose about it. 
Also, I am running on Pi 02w and I'm not sure if the dockerfile takes into consideration the steps your highlighted in the readme

manio, 2024-11-05:
Hi!
So... from the start:
> I haven't managed to uninstall aawgd but I did drop the built aa-proxy-rs onto the same path as you had and I chmod +x the binary to get it to run

Very good. That was the easiest way (to just replace aawgd) if you don't know how to make a startup script.

> I then get Bluetooth error: Bluetooth operation not permitted: UUID already registered. I suspect it's because I haven't removed aawgd.

Yes, it was rather because the `aawgd` was still running at the same time (or maybe even the bluetoothd has this still registered).
> 
> I also noticed the binary is not persisted on SD card.

This was figured out, so OK...

> I apologize for so many comments but I thought this would be a good place to be verbose about it.

This is OK. You can also edit your posts on github if you like.

> Also, I am running on Pi 02w and I'm not sure if the dockerfile takes into consideration the steps your highlighted in the readme

I also have the pi02w and the Dockerfile creates an image exactly for this architecture. So you're OK with this.

Regarding the log/problem:
The log is cut exactly in the place where the phone is changing wifi network, then it should start android auto but there has to be some other problem (there was two timeouts in the middle so maybe the further timing gets drifted too much to make a successful connection)...
Unfortunately I cannot tell you more without the logs (you can also grab it later).

You could also add eg `-s5` in the startup script:
/etc/init.d/S93aawgd
like this:
`start-stop-daemon -S -b -q -m -p "$PIDFILE" -x "/usr/bin/$DAEMON" -- -s5`
Then you should have transfer debug like here:
https://asciinema.org/a/686949

Some questions:
I assume it was working fine with aawgd for you?
Was you using AAWG_CONNECTION_STRATEGY=1. As this is the only one currently supported.
Can you try to connect several times?

Tttyagi123, 2024-11-05:
For people like me, who dont have a clue bout coding or scripting, is it possible to provide a flashable build with this mod incorporated in it?

NabeelUppel, 2024-11-05:
> Regarding the log/problem: The log is cut exactly in the place where the phone is changing wifi network, then it should start android auto but there has to be some other problem (there was two [...]

Thanks for the reply. 

So aawgd worked initially for me and then one day it just stopped and I never managed to get it working since. I believe I was using phone first connection strategy too. 
I did try several times and it always fails at the same place. I noticed from your demo video mine fails to says "Bluetooth Sequence Completed Successfully" so it fails before then.
It's also strange that the SSH session closes (I SSH using my laptop so I don't expect the pi's WiFi to disconnect like it did).
Where are aa-proxy-rs logs stored?

manio, 2024-11-05:
@Tttyagi123 
> For people like me, who dont have a clue bout coding or scripting, is it possible to provide a flashable build with this mod incorporated in it?

will try to just grab the official image and "inject" it for you... :)

@NabeelUppel
> Where are aa-proxy-rs logs stored?

cat /var/log/aa-proxy-rs.log

manio, 2024-11-05:
@Tttyagi123 
Here you are: https://skyboo.net/temp/aa/raspberrypizero2w-sdcard.img
btw. to be clear: I hope you read the README - it is still in a early development/beta phase - don't expect that it will just work...

Tttyagi123, 2024-11-05:
> @Tttyagi123 Here you are: https://skyboo.net/temp/aa/raspberrypizero2w-sdcard.img btw. to be clear: I hope you read the README - it is still in a early development/beta phase - don't expect that it will just work...

Sure, i also wanted to just try and see how it goes. Will give feedback after trying. Thanks a bunch 🙏

NabeelUppel, 2024-11-05:
@manio 
So it seems as though when it switches from default to accessory gadget the Pi loses power, I connected a powerbank to the power port and the ssh session is kept active but still does not work.

Starting to proxying data between TCP and USB...
and it hangs.

[aa-proxy-rs.log](https://github.com/user-attachments/files/17628872/aa-proxy-rs.log)

manio, 2024-11-05:
@NabeelUppel 
Can you try this executable?
https://skyboo.net/temp/aa/aa-proxy-rs

NabeelUppel, 2024-11-05:
The Pi doesn't lose power anymore but still hangs at
Starting to proxying data between TCP and USB...

[aa-proxy-rs.log](https://github.com/user-attachments/files/17629139/aa-proxy-rs.log)

manio, 2024-11-05:
Give me a moment ... I'll prepare a version with more debug...
In the mean time please just edit your startup line in the script to this:
```start-stop-daemon -S -b -q -m -p "$PIDFILE" -x "/usr/bin/$DAEMON" -- -d -s5```

manio, 2024-11-05:
Ok ... it's in the same place:
https://skyboo.net/temp/aa/aa-proxy-rs

NabeelUppel, 2024-11-05:
Will give this a try asap. 
When you say startup line in the script where exactly is that located or is that the aa-proxy-rs binary (I doubt it)

manio, 2024-11-05:
`/etc/init.d/S93aawgd`, same as above.

NabeelUppel, 2024-11-05:
Should I also change the DAEMON to be aa-proxy-rs or keep it as aawgd?

NabeelUppel, 2024-11-05:
> Ok ... it's in the same place: https://skyboo.net/temp/aa/aa-proxy-rs

I tried this but again the pi restarts.
[image]

I then used a power bank while connected to my phone and I get this:
[image]

manio, 2024-11-05:
@NabeelUppel 
Please try again - new version is there...
ps. your last log is cut at the place where it is most interesting for me :)

Please do it as always but with power bank, and provide me full logs - especially after the line "Starting to proxying"

I really don't know what you're doing there ;)
I would just do it like this:
0. make a copy of aawgd: cp /usr/bin/aawgd /usr/bin/aawgd-backup
1. Place aa-proxy-rs in its place (copy as /usr/bin/aawgd)
2. Change the line in the startup script (this is only enabling statistics and debug)
3. reboot - and you're done - it should work (for the first time)

...then try connect with AA - give it some time ofc; and then you can disconnect AA (if connected and you're doing it from phone) and connect afterwards to the AP via ssh and fetch the /var/log/aa-proxy-rs logs for me.

NabeelUppel, 2024-11-05:
Is there anyway to capture the start request by the car? 
I feel like I'm getting a lot of false positives, as in I connected the pi to power in my house and started the script and I recieved a accessory start request when I shouldn't have receive one seeing that it was not connected to my car

NabeelUppel, 2024-11-05:
> @NabeelUppel Please try again - new version is there... ps. your last log is cut at the place where it is most interesting for me :)
> Nothing happened below that, it hung at that message for about 2 mins and I disconnected. The message below was "Connection reset by peer" after I pulled power.

> Please do it as always but with power bank, and provide me full logs - especially after the line "Starting to proxying"
> Will try that now. 

> I really don't know what you're doing there ;) I would just do it like this: 0. make a copy of aawgd: cp /usr/bin/aawgd /usr/bin/aawgd-backup
> 
> 1. Place aa-proxy-rs in its place (copy as /usr/bin/aawgd)
> 2. Change the line in the startup script (this is only enabling statistics and debug)
> 3. reboot - and you're done - it should work (for the first time)
> Does this make it so that aa-proxy-rs starts on boot?
> ...then try connect with AA - give it some time ofc; and then you can disconnect AA (if connected and you're doing it from phone) and connect afterwards to the AP via ssh and fetch the /var/log/aa-proxy-rs logs for me.

manio, 2024-11-05:
I thought you are doing it in your car!
Read my last message.

NabeelUppel, 2024-11-05:
> I thought you are doing it in your car! Read my last message.

I am, but there was one time I tried it disconnected from my car and yet I still received the start accessory request

NabeelUppel, 2024-11-05:
I could not access the logs when running it on startup.

Without Powerbank:
[image]

With Powerbank:
[image]
Note: I pulled the power after about 2 mins

> @NabeelUppel Can you try this executable? https://skyboo.net/temp/aa/aa-proxy-rs

What did you change in this excutable to stop it from crashing?

manio, 2024-11-05:
Thank you for testing!

> What did you change in this excutable to stop it from crashing?

Nothing - the only thing changed was a slightly modified sequence and timing - starting USB to car later.
I still doesn't get why it was working with aawgd and now it can't with the same default strategy.
I can also see that USB is the problem here - first it is trying hard to switch to accessory, but when it finally happened, it's too late for android phone because it is disconnecting.

Ok ... I am providing you the version which is doing the opposite - trying to start USB much earlier...
please give it another try

Two things:
1. Why are you still starting it manually? Can you just do it as I described earlier? Put it to auto start via script?
2. I hope you don't have aawgd in the same time running?
3. Are you able provide me aawgd logs where it was working OK?

Also can you please describe how are you testing it exactly?
Please tell me in point like:
- I am powering it up,
- connecting usb to car
- logging via ssh
- etc...

manio, 2024-11-05:
OK... I've got it.
If your pi is crashing then I suspect something wrong here. It is not normal. My pi is not crashing, I can login in whenever it has the power, so I can fetch the logs eg. after AndroidAuto connects and even during it (if doing it from laptop).
I would try check power quality (it's probably from your car) or maybe add some cooling/radiators on top of CPU?

> That was like 3 versions and 5 months ago.
Maybe you can check it again, this could be valuable information. Maybe it is some regression. Here you have all releases:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/releases
Then we could be able to analyze what has changed.

NabeelUppel, 2024-11-05:
> OK... I've got it. If your pi is crashing then I suspect something wrong here. It is not normal. My pi is not crashing, I can login in whenever it has the power, so I can fetch the logs eg. after [...]

I'm starting to suspect that too,  I don't have any coolers or radiators pull additional just the power just a plain old RPi 02w. What I don't understand that it did work and absolutely nothing else changed when it just broke.

> > That was like 3 versions and 5 months ago.
> 
> Maybe you can check it again, this could be valuable information. Maybe it is some regression. Here you have all releases: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/releases Then we could be able to analyze what has changed.

I tried all the releases even the one I know worked for a while v0.3.

manio, 2024-11-05:
It may be many things including something on your phone...
Do you have some other phone to test?
Is it working if you connect it directly via USB cable?

I think that Pi restarting may be a sign of problems - maybe something is overheating and causing issues...

NabeelUppel, 2024-11-05:
> Ok ... I am providing you the version which is doing the opposite - trying to start USB much earlier... please give it another try

This worked a bit better, it did not crash and was able to use the auto start script and managed to pull the logs but my phone just flashed and went back to look for android auto.
Here are the logs
[aa-proxy-rs.log](https://github.com/user-attachments/files/17637478/aa-proxy-rs.log)

Tttyagi123, 2024-11-06:
> @Tttyagi123 Here you are: https://skyboo.net/temp/aa/raspberrypizero2w-sdcard.img btw. to be clear: I hope you read the README - it is still in a early development/beta phase - don't expect that it will just work...

Just brilliant!!!!
Earlier : if the pi connected within few seconds of booting up, it would work and if due to some reason it did not ( bt being off, wifi not connected to dongle, etc.) it would not connect until cable is plugged out from pi and re inserted. 

Now : i deliberately left it disconnected initially, went on to drive, connected bt afterwards and it connected instantaneously!!! 

This is just great!!! Thanks 👍

Tttyagi123, 2024-11-06:
> > Ok ... I am providing you the version which is doing the opposite - trying to start USB much earlier... please give it another try
> 
> This worked a bit better, it did not crash and was able to use the auto start script and managed to pull the logs but my phone just flashed and went back to look for android auto. Here are the logs [aa-proxy-rs.log](https://github.com/user-attachments/files/17637478/aa-proxy-rs.log)
> 
> Edit: I lie, I realised I didn't plug it into the otg port when I tested the above. After fixing it, it crashes again.
> 
> Something interesting I noticed that everytime I connect to Bluetooth my headunit throws a USB error so I think there's a miscommunication between the two. I have a Mazda 3 2015 with mzd connect.

I always get  usb error too but it connects fine afterwards, could be something else....

NabeelUppel, 2024-11-06:
> @NabeelUppel
> 
> > Edit: I lie, I realised I didn't plug it into the otg port when I tested the above. After fixing it, it crashes again.
> 
> If I understand correctly the OTG port is for communication but also Pi can be powered with it (like I use it), but there is the other port for power - I assume when you plug your power bank there then it is not crashing?
> 
> > Something interesting I noticed that everytime I connect to Bluetooth my headunit throws a USB error so I think there's a miscommunication between the two.
> 
> As @Tttyagi123 said - not necessarily - I also have such error on my car screen. It is rather because the raspberry Pi is enabling default gadget, waiting for notice and doing sequence of gadget switching - this is just probably noticed by the car and I think this is correct. I analyzed your logs and the problem is quite different: The Pi two times timeouts with `Timeout waiting for accessory start, trying to recover...` But finally when it manages to successful switch: `Switched from default to accessory gadget` And your android phone connected OK, then it is not starting any transmission stuck at: ` (1) aa_proxy_rs::io_uring: USB: before read` Ten seconds later android is disconnecting because of no data. And on your phone you stay with spinning "Looking for android auto".

I've ordered one of those Y Usb cables to give it more from my second USB port, it's gonna be a while before it arrives. Thank you for all the help, I'll continue to play around with it and if I make any breakthrough I will definitely report back!

> I was also asking if it is working if you connect your phone directly...

Android Auto works perfectly fine connecting my phone via cable.

manio, 2024-11-06:
> Android Auto works perfectly fine connecting my phone via cable.

I prepared another version - increased one timeout between gadget switching - you can give it a try if you want.

Tttyagi123, 2024-11-06:
> @NabeelUppel
> 
> > Edit: I lie, I realised I didn't plug it into the otg port when I tested the above. After fixing it, it crashes again.
> 
> If I understand correctly the OTG port is for communication but also Pi can be powered with it (like I use it), but there is the other port for power - I assume when you plug your power bank there then it is not crashing?
> 
> > Something interesting I noticed that everytime I connect to Bluetooth my headunit throws a USB error so I think there's a miscommunication between the two.
> 
> As @Tttyagi123 said - not necessarily - I also have such error on my car screen. It is rather because the raspberry Pi is enabling default gadget, waiting for notice and doing sequence of gadget switching - this is just probably noticed by the car and I think this is correct. I analyzed your logs and the problem is quite different: The Pi two times timeouts with `Timeout waiting for accessory start, trying to recover...` But finally when it manages to successful switch: `Switched from default to accessory gadget` And your android phone connected OK, then it is not starting any transmission stuck at: ` (1) aa_proxy_rs::io_uring: USB: before read` Ten seconds later android is disconnecting because of no data. And on your phone you stay with spinning "Looking for android auto".
> 
> I was also asking if it is working if you connect your phone directly...
> 
> @Tttyagi123 Thank you for your feedback! I'm glad it is better and regarding this:
> 
> > Edit : on second attempt I wasn't pucky though, it stayed disconnected :(
> 
> If you are talking about connecting after: disconnection/out of range/etc: I still have to implement this (it is on my todo list), but if you are talking about initial connection (for the first time since bootup) - I cannot tell you more without your logs :(

M sure soon you will be able to make it disconnect and reconnect without issues too, cheers! :)

manio, 2024-11-07:
@NabeelUppel 
> I prepared another version - increased one delay between gadget switching - you can give it a try if you want.

Did you test it? As I have more ideas :wink:

manio, 2024-11-07:
@NabeelUppel 
I am slowly ending my day so if you're at this, please try these two binaries:
1. https://skyboo.net/temp/aa/aa-proxy-rs
2. https://skyboo.net/temp/aa/aa-proxy-rs2

NabeelUppel, 2024-11-07:
> @NabeelUppel I am slowly ending my day so if you're at this, please try these two binaries:
> 
> 1. https://skyboo.net/temp/aa/aa-proxy-rs
[image]

> 2. https://skyboo.net/temp/aa/aa-proxy-rs2
[aa-proxy-rs.log](https://github.com/user-attachments/files/17664341/aa-proxy-rs.log)

No dice with either unfortunately. Hoping to get a y-usb cable to give the pi more juice soon.

manio, 2024-11-07:
Ok, thanks. I was hoping that second version would work...

manio, 2024-11-09:
@Tttyagi123 
Still using it?
I know you was connecting manually to the bluetooth. Now it can be configured to automatically connect to your phone (as the original `aawgd` was doing by default). I added this today.
There should be also faster connection (and maybe even no USB disconnect on your car headunit) as I optimized it a lot.

Tttyagi123, 2024-11-09:
> @Tttyagi123 Still using it? I know you was connecting manually to the bluetooth. Now it can be configured to automatically connect to your phone (as the original `aawgd` was doing by default). I [...]

Ofcourse I am, where is the link, m dieng to try it, u make any changes in future, give me a shout plz i wil love to test :)

manio, 2024-11-09:
You want the full image like we do recently or the binary is ok?

Tttyagi123, 2024-11-09:
> You want the full image like we do recently or the binary is ok?

Unfortunately m illiterate in terms of binaries lol i just know to flash images, if u may plz. M embarrassed of it but it is what it is :D

manio, 2024-11-09:
https://skyboo.net/temp/aa/raspberrypizero2w-sdcard2.img

Tttyagi123, 2024-11-09:
> https://skyboo.net/temp/aa/raspberrypizero2w-sdcard2.img

Thnks a bunch mate, cheers!

manio, 2024-11-09:
It's a fresh image so you probably need to pair it once, then it should auto-reconnect on every reboot.

Tttyagi123, 2024-11-09:
> It's a fresh image so you probably need to pair it once, then it should auto-reconnect on every reboot.

Gonna test it tomorrow and let u know how it goes

Tttyagi123, 2024-11-10:
[image]

manio, 2024-11-10:
Ok... thanks. Maybe the old gadget mode is needed in your car. I prepared it again (same link).

Tttyagi123, 2024-11-10:
> Ok... thanks. Maybe the old gadget mode is needed in your car. I prepared it again (same link).

Hurraayyyyy!!!!!!
[image]

manio, 2024-11-10:
I only did first connect when you enter your card (and Pi is booting).

> i disconnected and waited for it to reconnect but that din happen so i tried to connect it manually but that din work either. I tried it several times during my short drive and it failed each time to connect so far.

I know - this is not yet done...

Tttyagi123, 2024-11-10:
> I only did first connect when you enter your card (and Pi is booting).
> 
> > i disconnected and waited for it to reconnect but that din happen so i tried to connect it manually but that din work either. I tried it several times during my short drive and it failed each time to connect so far.
> 
> I know - this is not yet done... but if it doesn't connect and you cannot connect manually (for the first time) than this is strange. Logs would be useful but I know you're not familiar with ssh...

Ssh is something i should be able to do if told how to, is there any guide for it?

manio, 2024-11-10:
[Here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#getting-logs) is a guide.
You just need to login there if there was some problem and then use this command:
```cat /var/log/aa-proxy-rs.log```
To obtain the logs. But you have to take a screenshot or save whole session to file if you're doing it from phone.

Tttyagi123, 2024-11-10:
> [Here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#getting-logs) is a guide. You just need to login there if there was some problem and then use this command: `cat [...]

Will try it n share the log with u

manio, 2024-11-11:
Yes, this is expected behavior, but stay tuned for reconnect code...

Tttyagi123, 2024-11-11:
@manio 
Point 1 is true but point 2 is false in my case, will check if i can get my hands on the logs

manio, 2024-11-11:
Hold on @Tttyagi123 ... I will have the reconnect code in a short while...

[68 more replies]

commit b6c292f0 (2024-11-25): docker: prepare SD card images, update README.md

Closes #1
closed 2024-11-25 by manio