aa-proxy/aa-proxy-rs #3: segmentation fault

Ioniq3, 2024-12-06:
HI manio,

Great work, it works really good at rpi3a.

But in rpi0w aa-proxy-rs says "segmentation fault" and it doesn't work

manio, 2024-12-06:
Thanks for the feedback!
So you just used the ready image for `rpi0w`?
Does it throw the segfault right on the start (no single line of log)?

I am using it with `Pi Zero 2 W` and unfortunately don't have a `0w`.
But I'll try to figure the problem out if I could...

Ioniq3, 2024-12-07:
The cpu of the rpi0w only support 32bit instruction set, no A53 arquitecture, so maybe it must be the problem.
How can I build for arm1176jzf, as I see this is the main differente betwenn raspberrypi0w_defconfig and aspberrypizero2w_defconfig

Thanks

manio, 2024-12-07:
The build is 32-bit:
`ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV)`
built on `arm-unknown-linux-gnueabihf` target,
so this must be some other problem I think...

manio, 2024-12-07:
I found this: https://github.com/mdaffin/rpizw-rover/issues/2:

> Using those instructions, even hello world segfaults for me when compiled from Ubuntu for Pi Zero (though the same binary runs fine on armv7 like RPi3).

Sound familiar? ;)
It might be about some ubuntu problem compiling... I'm still investigating.
Are you able to place ELF only on your target Pi to see if it works if I have something for test?

manio, 2024-12-07:
@Ioniq3 Please try this: https://skyboo.net/temp/aa/aa-proxy-rs

Ioniq3, 2024-12-07:
Hi manio,

I followed the following https://github.com/locnnil/rust_cross_compilation

and now works, I can build for rpi0w, rpi-tools are necessary instead of arm-linux-gnueabi-gcc

manio, 2024-12-07:
This is what I did to create the mentioned binary. I'd be grateful if you confirm that it is also working :)

manio, 2025-01-10:
No problem, thanks for the feedback. I'll leave this open as I have to fix the build for this platform.

commit b86e87df (2025-04-17): Docker: fix building `pi zero w` image

Closes: #3
closed 2025-04-17 by manio