aa-proxy/aa-proxy-rs #7: ConfigFs path does not exist

iglooom, 2024-12-27:
Hi, I'm trying to tweak the existing AA Wireless adapter, which is based on AllWiner V851SE + AIC8800 Wifi/BT and Tina Linux 4.9.191
I've compiled aa-proxy-rs as static with `RUSTFLAGS='-C target-feature=+crt-static' cargo build --release`
And it can start on this adapter but exits with this
```root@WirelessAADongle:~$ aa-proxy-rs -d
1970-01-01, 01:03:24.765 [ERROR]  main:  Error creating/opening log file: "/var/log/aa-proxy-rs.log": Os { code: 2, kind: NotFound, message: "No such file or directory" }
1970-01-01, 01:03:24.766 [WARN]  main:  Will do console logging only...
1970-01-01, 01:03:24.767 [INFO] 🛸 aa-proxy-rs is starting, build: 20241227_141320, git: 20241224-38a633c
1970-01-01, 01:03:24.768 [INFO]  main:  📜 Log file path: /var/log/aa-proxy-rs.log
1970-01-01, 01:03:24.768 [INFO]  main:  ⚙ Showing transfer statistics: disabled
thread 'main' panicked at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.5.0/src/lib.rs:148:48:
called `Result::unwrap()` on an `Err` value: Os { code: 38, kind: Unsupported, message: "Function not implemented" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
1970-01-01, 01:03:24.776 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-01, 01:03:24.777 [ERROR]  main:  🔌 USB init error: ConfigFs path does not exist
1970-01-01, 01:03:24.779 [INFO]  bluetooth:  🥏 Bluetooth alias: WirelessAADongle
1970-01-01, 01:03:24.780 [ERROR]  main:  Bluetooth error: internal error: join error: task 4 was cancelled
1970-01-01, 01:03:24.781 [INFO]  main:  Trying to recover...
```
While the path present in the system
```
root@WirelessAADongle:~$ ls -l /sys/kernel/config/
drwxr-xr-x    3 root     root             0 Jan  1 00:11 usb_gadget
root@WirelessAADongle:~$ mount
/dev/root on /rom type squashfs (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=27428k,nr_inodes=6857,mode=755)
/proc on /proc type proc (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
sys on /sys type sysfs (rw,relatime)
/dev/by-name/UDISK on /mnt/UDISK type ubifs (rw,relatime)
overlayfs:/mnt/UDISK/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/mnt/UDISK/overlay/upper,workdir=/mnt/UDISK/overlay/workdir)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
none on /sys/kernel/debug type debugfs (rw,relatime)
none on /sys/kernel/config type configfs (rw,relatime)
adb on /dev/usb-ffs/adb type functionfs (rw,relatime)
```
Initially `/sys/kernel/config/usb_gadget/` is empty and for example adbd service starts like this
```
[ -d /sys/kernel/config/usb_gadget ] || {
        mount -t configfs none /sys/kernel/config
        mkdir /sys/kernel/config/usb_gadget/g1
        echo "0x18d1" > /sys/kernel/config/usb_gadget/g1/idVendor
        echo "0xD002" > /sys/kernel/config/usb_gadget/g1/idProduct
        mkdir /sys/kernel/config/usb_gadget/g1/strings/0x409
        echo "Allwinner" > /sys/kernel/config/usb_gadget/g1/strings/0x409/manufacturer
        echo "Tina" > /sys/kernel/config/usb_gadget/g1/strings/0x409/product
}
[ -d /sys/kernel/config/usb_gadget/g1/configs/c.1 ] || {
        mkdir /sys/kernel/config/usb_gadget/g1/configs/c.1
        echo 0xc0 > /sys/kernel/config/usb_gadget/g1/configs/c.1/bmAttributes
        echo 500 > /sys/kernel/config/usb_gadget/g1/configs/c.1/MaxPower
        mkdir /sys/kernel/config/usb_gadget/g1/configs/c.1/strings/0x409
}
[ -d /sys/kernel/config/usb_gadget/g1/functions/ffs.adb ] || {
        mkdir /sys/kernel/config/usb_gadget/g1/functions/ffs.adb
}
[ -e /sys/kernel/config/usb_gadget/g1/configs/c.1/ffs.adb ] || {
        ln -s /sys/kernel/config/usb_gadget/g1/functions/ffs.adb/ /sys/kernel/config/usb_gadget/g1/configs/c.1/ffs.adb
}
[ -d /dev/usb-ffs/adb ] || {
        mkdir /dev/usb-ffs
        mkdir /dev/usb-ffs/adb
        mount -o uid=2000,gid=2000 -t functionfs adb /dev/usb-ffs/adb/
}
get_serialnumber
echo $SERIALNUMBER > /sys/kernel/config/usb_gadget/g1/strings/0x409/serialnumber
```

Do you have any idea how to start aa-proxy in this system?

manio, 2024-12-27:
Hi, what kernel version are you using?

iglooom, 2024-12-27:
```
root@WirelessAADongle:~$ uname -a
Linux WirelessAADongle 4.9.191 #11 PREEMPT Fri Dec 27 16:25:58 UTC 2024 armv7l GNU/Linux
```

iglooom, 2024-12-27:
I've tried to build it with `armv7-unknown-linux-gnueabi` but no difference

manio, 2024-12-27:
This soft is using tokio-uring and according to the docs: `The library requires Linux kernel 5.10 or later.`
So unfortunately your kernel is too old :(

iglooom, 2024-12-27:
Got it, thanks for this info.

closed 2024-12-27 by iglooom

manio, 2024-12-29:
@iglooom 
Just by curiosity: is it hard to upgrade the kernel to more recent or this is not upgraded because of the hardware?