aa-proxy/aa-proxy-rs #9: Disconnecting -> rpi 3b <-> rpi 0 2W

trybikowski, 2024-12-30:
Hi,
I’m having a connection issue. I installed the latest release (from yesterday) and I see something like this on the screen:
[image] - connected / disconnected.
I’m using a Raspberry Pi 0 2W, which is connected to a Raspberry Pi 3 Model B. I’ve tried both OpenAuto Pro 16.1 and Crankshaft. Both systems are showing the same issue. Android Auto is updated to the latest version.

trybikowski, 2024-12-30:
The previous version didn’t work either. I thought the problem was with the 'aa-proxy-rs' implementation, but it’s not. :) 
Okay, I’ll try asking @KreAch3R. 
Thanks a lot!

KreAch3R, 2024-12-30:
Hello,

What are you trying to do? 

Just use aa-proxy-rs with WirelessAndroidAutoDongle and the instructions of this repo?

Or followed my repo aa-proxy-rs at any part of it?

Does WirelessAndroidAutoDongle work for you?

trybikowski, 2024-12-30:
Hi,
I'm trying to establish a wireless connection between my phone and the RPi3 in my car :). With a wired connection, everything works fine on Crankshaft, but it doesn't work with OpenAuto (OpenAuto does not support the latest AA version, and it's not possible to open AA with an older version).
I'm using this setup:
[image]
I inserted the SD card into the RPi Zero 2W and connected it to the RPi3 running OpenAuto Pro and Crankshaft. Unfortunately, it does not work.

KreAch3R, 2024-12-30:
Does the image of WirelessAndroidAutoDongle work for your setup?

trybikowski, 2024-12-30:
No, it doesn't work, and I was hoping you could help me with this :).
As I mentioned in my first comment, the wireless dongle keeps getting "disconnected" and "connected" repeatedly by the head unit (RPi 3).

Do you have any suggestions or advice on how to resolve this issue?

PS
I’ve tried all the images from @manio and nisargjhaveri.

manio, 2024-12-30:
My suggestion is to grab logs but from the source Pi02W and provide me this :)

manio, 2024-12-30:
@trybikowski /var/log/aa-proxy-rs.log is what we'd like to see :)
You need to log in eg via ssh first (or with a keyboard), refer this: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#getting-logs

trybikowski, 2024-12-30:
I have logs from OpenAuto and Crankshaft (it wasn't easy for me to get them :)). Unfortunately, I am still facing the same issue—disconnecting.
Sadly, my Raspberry Pi Zero 2W broke after the tests (it shows no response after being powered on). I will need to make a complaint to the seller, so further tests won't be possible for now. 
However, maybe you’ll be able to say something based on the logs.

My equipment:
Phone: Pixel 8 with Android 15
AA: latest version 13.3.644654
Raspberry 3 ver. B
Wifi dongle: Raspberry zero 2 W

[crankshaft.odt](https://github.com/user-attachments/files/18277495/crankshaft.odt)
[openauto.odt](https://github.com/user-attachments/files/18277497/openauto.odt)

KreAch3R, 2024-12-30:
What you are calling openauto logs is actually aa-proxy-rs.log. I think it shows that the phone gets connected, AA starts, but then it stops abruptly. I think the disconnection issue and your hardware problem may be related, are you sure that the RPI3 provides enough power through USB to the rpi2w? I used a powered usb hub to connect them.

manio, 2024-12-30:
Oh... why you saved it in LibreOffice instead of pure log/txt? ;)

Looking at crankshaft.odt: It seems that Pixel phone has successfully connected after several tries but the problem was that the car/whatever you have has not started trasmision in a timely manner - it was a USB timeout (it is set to 10 seconds). It has to start transmission with a frame.
While looking at openauto.odt: there is the opposite situation - the TCP (phone?) is not responding with data and you have TCP read timeout.

As I said I am not familiar with those systems (crankshaft, openauto) so maybe @KreAch3R will be more proper person here :)
Anyway if your Pi is screwed than nothing we can do now...

KreAch3R, 2024-12-30:
Yes, I read those the same way. The only way for the USB transmission to start but end abruptly on my tests, was that I killed the openauto app on the Host rpi and then tried to launch the script (noone was listening on the other end), and some random issues which always get fixed with a restart. 

But the most reproducible was that OpenAuto/Crankshaft was not running correctly/was not running whatever service they run to accept USB AA connections / wasn't running at all.

trybikowski, 2024-12-30:
@KreAch3R -  you might be right (not enough power for the Pi Zero 2 W). I connected it directly to the RPi3, and it’s possible that it got damaged because of that.
@manio - thanks for the response. Is it possible to extend the 10-second timer? (I have no idea abput that :))

When I get a new Pi Zero 2 W, I’ll get back to you.

manio, 2024-12-30:
This timeout is way way more then needed, but I'll think about some config switch for this.

manio, 2024-12-30:
Done in https://github.com/manio/aa-proxy-rs/commit/7282f4ce64d0436ccd645f771f3f543cbd257702