aa-proxy/aa-proxy-rs #11: AA on head unit appears and disappear after short time

Icobob, 2025-01-06:
Manio first tank you very much your aa-proxy! I first install v0.2.0. It worked very stable on my raspberry 3a+ 
I decided to go to new v0.3.0 and here start the problem. AA started appear on head unit and after short time disappear and again started and appear and suddenly disappear. I tried  back to v 0.2.0 - same result.
What could be the problem?

manio, 2025-01-06:
Thanks @Icobob !
If it works and stopped - first try to forget the bluetooth device on your phone (may be also necessary on your car head unit) - if this doesn't help I need to see what is going on... so please provide me the logs. You can find information how to obtain it in the README.

manio, 2025-01-06:
According to the logs all is fine... The bluetooth handshake was done properly, a proxy connection has been established and it just should transfer data. Moreover it should timeout and retry if there is no data. Maybe you grab the logs too early?

Can you edit the startup [parameters](https://github.com/manio/aa-proxy-rs/blob/main/contrib/S93aa-proxy-rs#L10) and try to change `-s60` to `-s5` - to have statistics every 5 seconds?

Maybe you can also try legacy mode (`-l`) but if it was working before I doubt this is the case.

Icobob, 2025-01-06:
Changed -s60 to -s5 and wait longer. Attached logs
[logs.zip](https://github.com/user-attachments/files/18322999/logs.zip)

manio, 2025-01-07:
Thanks, now please try to increase timeout as it seems this is triggering the disconnection. Test it eg with a minute to see if it help:
Add a `-t60` parameter to the same file.

Icobob, 2025-01-08:
Armament -t make does not help. I tried several values including default -t5.  No log aa-proxy-rs. I suspect -t is killing §§-proxy-rs
[Logs.zip](https://github.com/user-attachments/files/18349084/Logs.zip)

manio, 2025-01-08:
Ok... I'll check that parameter myself soon...
btw: you closed this issue by accident?

Icobob, 2025-01-08:
Yes by accident.

manio, 2025-01-09:
Ok... Tested and it is working for me but I think I know where is the problem. You are using stable version 0.3 while I am telling about recent version. When you pass this argument it is exiting with error :)

Can you please try recent version?
This one: https://github.com/manio/aa-proxy-rs/actions/runs/12557287185/artifacts/2373679658

Icobob, 2025-01-09:
Thank you. With -s5 -t30 The behavior is as I reported in my first message. My be AA stay on head unit screen a bit longer. During attempts to transfer logs to laptop WiFi disconnected several time. 
[Logs.zip](https://github.com/user-attachments/files/18362749/Logs.zip)

manio, 2025-01-09:
I can see in the logs the connection for about two minutes. All seems fine - at least to the end of the log...
So is it now better/working?
I am confused...

Icobob, 2025-01-09:
Duration of last test was around 15 minutes. Several time AA appears on head unit screen stays sometimes longer sometimes very short  and then disappears. With Winscp I transfer logs to laptop but WiFi connection breaks several time and I must reconnect. Is it normal?

manio, 2025-01-10:
> Duration of last test was around 15 minutes

So it was working OK for this 15 minutes? Is it working fine if you connect your phone directly to car via USB cable?

> With Winscp I transfer logs to laptop but WiFi connection breaks several time and I must reconnect. Is it normal?

Yes... but only when raspberry is trying to connect to your phone via BT (and searching longer for the device). BT and WiFi are interfering in this case.

Icobob, 2025-01-10:
Thank you very much for you efforts! I thing the problem is in my mobile or head unit, As I said before it was working without problem with your previous stable version. So I will investigate deeply and come back.