aa-proxy/aa-proxy-rs #12: Security Enhancement Requests

liamfleming26, 2025-01-21:
First, let me sincerely thank you, @manio, for creating and maintaining this project! It's a handy tool, and your work is greatly appreciated.

If feasible, I wanted to propose a couple of security enhancements that might improve the system's safety and usability.

## 1. Bluetooth Pairing: Fixed Pairing Code
Currently, Bluetooth pairing is open-ended, allowing any nearby device to initiate a connection. This can lead to unauthorised devices pairing with the system.

Would it be possible to implement a fixed pairing code that users can configure (e.g., through an environment variable or configuration file)? This would ensure that only devices with the correct code could connect to the Raspberry Pi.

I understand that Bluetooth pairing mechanisms can vary depending on the implementation and Linux system settings, and I am not deeply familiar with how Bluetooth is handled in this project. If guidance or testing is needed, I'd be happy to help where I can.

## 2. WiFi Connection: Improved Security
The current setup reads the SSID and WPA2 PSK from the /etc/hostapd.conf file, which works well for setting up a wireless access point. However, this approach might leave the system vulnerable if the credentials are static or predictable.

### **Two possible solutions to enhance WiFi security:**
#### **Randomised SSID and WPA2 PSK on Boot**
Randomising the SSID and PSK on each boot would reduce the risk of unauthorised access to the device. This would then be picked up when the aa-proxy-rs daemon is started.

#### **WiFi Direct Connection (Peer-to-Peer)**
Alternatively, would establishing a WiFi Direct connection between the Raspberry Pi and the Android device be more secure and practical? This would eliminate the need for the Raspberry Pi to broadcast an SSID continuously and rely on WPA2 for security. WiFi Direct could provide a more direct and secure connection with reduced exposure to unauthorised access.

I understand these enhancements might involve significant changes to the current implementation, so I'd love to hear your thoughts on their feasibility and priority. If these align with the project's goals, I'd be happy to contribute ideas, test, or do anything else that could assist in their development.

Thank you again for all your hard work on this project!

manio, 2025-01-23:
> First, let me sincerely thank you, @manio, for creating and maintaining this project! It's a handy tool, and your work is greatly appreciated.

Much thanks, Liam

> Bluetooth Pairing: Fixed Pairing Code

Unfortunately I am not an bluetooth expert, but according to [this doc](https://www.ellisys.com/technology/een_bt07.pdf):
_Bluetooth pairing, based on a PIN-code, didn't provide any real level of security against sniffing_, maybe this is why I am nowadays not asked about PIN when pairing. I assume the pairing is now secure even without a PIN - as modern bluetooth pairing is doing it without it somehow.
I would just say that to make it more secure you can do the following: disable BLE advertisement and enable connection only to your phone MAC address - this is possible using command line switches. This way it should connect only to your specific phone. Besides the bluetooth adapter is visible only for short period of time after starting (when initiating the connection).

Regarding point two: as I am dependent on original Nisarg's project than this has to be covered by the [WirelessAndroidAutoDongle](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle) itself: If the generated password will land in `hostapd.conf`, then all should work fine.

Specifically:
> Randomised SSID and WPA2 PSK on Boot

This is probably covered [here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/pull/200)

> WiFi Direct Connection (Peer-to-Peer)

And this is covered [here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/243)

liamfleming26, 2025-01-27:
Thanks for the feedback there. I'll keep a close eye on the other repo too. I have a seperate issue with connectivity using the pi 0 2 w but I'll create a separate issue for that instead.

closed 2025-01-27 by liamfleming26