aa-proxy/aa-proxy-rs #14: Issue with Raspberry Pi Zero 2W Connection

liamfleming26, 2025-01-27:
## Description  

Hi Folks,  

Thanks for this fantastic project! I've been using it on a Raspberry Pi 4, and it has been working flawlessly. However, I've encountered an issue when trying to use the same setup with a Raspberry Pi Zero 2W.  

## Details  
I've flashed the appropriate image for the Zero 2W ([manio/aa-proxy-rs/releases/download/v0.3.0/raspberrypizero2w-sdcard.img.xz](https://github.com/manio/aa-proxy-rs/releases/download/v0.3.0/raspberrypizero2w-sdcard.img.xz)) using Balena Etcher v1.19.25. Unfortunately, I've only been able to successfully connect once, and subsequent attempts have failed.  

### What I've Tried  
- **SD Cards:** Tested multiple SD cards, including the one that worked perfectly with the Raspberry Pi 4.  
- **Cables:** I tested several micro-USB cables to ensure they support data transfer, using an Android phone to verify them.  
- **Image Redownload:** Redownloaded the image file to ensure no errors during the download process.  
- **Reflashing:** Reflashed the SD cards with the freshly downloaded image to eliminate potential issues with the original flash.  

Despite these efforts, the only result I can achieve is that the Android Auto icon appears on my car's head unit when using the Raspberry Pi Zero 2W. Conversely, the Raspberry Pi 4 boots in around 7 seconds and successfully connects with the head unit and phone.

## Steps to Reproduce  
1. Download the Raspberry Pi Zero 2W image and flash it using Balena Etcher.  
2. Insert the SD card into the Raspberry Pi Zero 2W.  
3. Connect the device to the car's head unit using a verified micro-USB cable.  
4. Attempt to establish a connection.  

## Expected Behaviour  
The Raspberry Pi Zero 2W should function similarly to the Raspberry Pi 4, enabling a full connection and seamless operation.  

## Actual Behaviour  
The only observable behaviour is the appearance of the Android Auto icon on the car's head unit.  

## Additional Information  
- **Phone Model:** Samsung Galaxy S22 Ultra  
- **Device:** Raspberry Pi Zero 2W  
- **Flashing Tool:** Balena Etcher  
- **Car Make:** Ford  
- **Car Model:** Focus  
- **Car Year:** 2020  
- **Car Head Unit:** Ford Sync 3  

I'll gather some logs from the Raspberry Pi Zero 2W and share them here as soon as possible. Please let me know if you need any specific details or tests to help identify the issue!  

Thanks again for your hard work on this project—it's much appreciated.

liamfleming26, 2025-01-27:
As a follow-up to my initial issue, I’ve tested using the `--legacy` flag as documented in the README. Unfortunately, this seemed to make things worse, as I couldn’t even get the Android Auto logo to display on the head unit.  

I’m now planning to experiment with the `--timeout` flag and try setting it to 10 seconds and 15 seconds to see if this helps maintain a successful connection. Once I’ve had a chance to test this, I'll report back with my findings.

manio, 2025-01-28:
Liam,
The `timeout` option indeed has been added but it is in not-yet-released tree. If you want to test it, just head to the recent build artifact for the moment, which is [here](https://github.com/manio/aa-proxy-rs/actions/runs/12557287185/artifacts/2373679658).
Inside the archive should be a working image for `pi zero 2 w`.

Test it with some varying timeout values and reply back. And as always - I cannot help you further with this if you will not provide me the aa-proxy-ls logs :)

manio, 2025-07-28:
@rickgitdone
I looked into your logs and indeed I can see various problems :(
The message "phone cannot connect to our WiFi AP" means that it is some WiFi problem - this generally means that your phone is unable to connect to raspberry pi WiFi for some reason. It is then replying back via Bluetooth with this info. All the time your phone was near the raspberry? Do you have a noise environment? What helps me in those cases are sometimes disabling WiFi+Bluetooth for a while and then start over. If the problem persist it is worth to try removing phone profile from the car, and car from the Android Auto settings, and forget Bluetooth profile.
One connections seems perfectly fine but it ended up with disconnection as well... I can also see some proper connection but then no data was sent (I can see it in stats) so it has been disconnected because of transfer stall.

Are you sure you have good quality USB cable? Raspberry Pi Zero 2 W is 2.4GHz only radio which sometimes interferes with Bluetooth during handshake phase. But on the other hand - it is working pretty reliable for me...
Let me know if you still have problems so maybe we could try to figure this out somehow...

manio, 2025-07-28:
Hmmm... is it better if you set:
`bt_timeout_secs = 3600` ?

rickgitdone, 2025-07-28:
> Hmmm... is it better if you set: `bt_timeout_secs = 3600` ?

I will test this.. Data transfer timeout is currently 30 .. should I adjust this as well?

manio, 2025-07-28:
> Data transfer timeout is currently 30 .. should I adjust this as well?

This applies to a working connection, so doesn't matter.

manio, 2025-07-28:
Did you try to enable/disable legacy mode?

rickgitdone, 2025-07-28:
# Enable legacy USB mode (some HeadUnits/cars needs this enabled for compatibility)
  legacy = true

was set ..

manio, 2025-07-28:
Ok... If you like I can prepare an optimized 64bit image for pi02w, but these works are currently ongoing...
if you can wait a day/two we could test it...

rickgitdone, 2025-07-30:
> Thanks I will test today

This 64 bit version connects and stays connected unlike the other version. I also turned off and restarted the vehicle and AA stayed connected ... good progress... slight blips however this one still has the wifi scanning on.

rickgitdone, 2025-09-10:
related to the ongoing stuttering on mainly the Pi 2W .. would using a Realtime OS be possible to reduce the stutter?

There is a kernel option to use RT OS  .. [Here](https://forums.raspberrypi.com/viewtopic.php?t=345326)