aa-proxy/aa-proxy-rs #15: Raspberry Pi 0w build broken

discojon, 2025-02-05:
Just a FYI, the RPi0w build seems to be broken.  I've flashed with RUFUS and RPi imager and multiple cards but the device will never boot.  I get no response from the headunit and never see the bluetooth device.  I'd like to try this build as i have intermittent stuttering with the main build.

manio, 2025-02-05:
This is a duplicate of https://github.com/manio/aa-proxy-rs/issues/3

Please try this image until I fix this: https://skyboo.net/temp/aa/raspberrypi0w-sdcard.img.xz

manio, 2025-02-06:
Can you please provide me the logs?

manio, 2025-02-06:
And also try the legacy mode (`-l` in options), maybe your car needs this, anyway I cannot tell you more without logs...