aa-proxy/aa-proxy-rs #17: No WiFi 5GHz

Zeragonii, 2025-02-24:
Hi,

Not sure if this is actually an issue or if I'm just not understanding things we'll, but from what I can tell my Pi Zero 2 W is only letting my phone (Pixel 9 Pro XL) connect via 2.4GHz @ 72mbps, the audio quality is in the dirt by proxy and I can't tell if it's something I've done wrong or if it's a known/unknown bug?

Can provide logs if needed

manio, 2025-02-25:
I am also using `Pi Zero 2 W` daily without any audio issues even with 2.4GHz WiFi. You can provide the logs if you like, I'll take a look.
If your question is about 5GHz and connectivity as a whole, it has been many times covered in the main project: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle

Zeragonii, 2025-02-25:
Okay I'll get my hands on the logs, but otherwise the TLDR is that the Pi Zero 2 W won't do 5GHz well/at all?

manio, 2025-02-25:
> TLDR is that the Pi Zero 2 W won't do 5GHz well/at all?

true

iceteaSA, 2025-03-17:
@manio Do you know if 5GHZ would cause less interference? I am running into an issue where it takes long to connect with the phone in my pocket, or it stutters or disconnects while I am driving. I notice a marked improvement when putting my phone right next to the pi20W, but that kinda defeats the point for me 😅  

I have already tried changing the Wi-Fi channel to 11, which seems to have helped, but that could just be placebo at work.

manio, 2025-03-18:
Exactly, and if you're at this, here you can find the info how to do it **properly**: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/245

iceteaSA, 2025-03-18:
Now I just need a pi with 5GHZ 😓