aa-proxy/aa-proxy-rs #18: Connection errors with zero2w

rohanb1216, 2025-02-25:
When trying to connect with aa-proxy-rs, the first time, the server starts but then closes due to timeout. From then onwards, server will close due to an IO error (system error 5). Using default config plus -a -k -t 15 --debug --legacy.

```
1970-01-01, 00:00:04.742 [34m[INFO][0m 🛸 [1m[34maa-proxy-rs[0m is starting, build: 20241231_082206, git: 20241231-cf49a7c
1970-01-01, 00:00:04.742 [34m[INFO][0m [3m[90m main: [0m 📜 Log file path: [1m[32m/var/log/aa-proxy-rs.log[0m
1970-01-01, 00:00:04.742 [34m[INFO][0m [3m[90m main: [0m ⚙️ Showing transfer statistics: [1m[34m1m[0m
1970-01-01, 00:00:04.776 [34m[INFO][0m [3m[90m usb: [0m 📬 Starting UEvent listener thread...
1970-01-01, 00:00:04.776 [36m[DEBUG][0m (5) aa_proxy_rs: Found interface: Interface { name: "wlan0", flags: 69699, mac: [216, 58, 221, 143, 64, 20], address: 10.0.0.1, scope_id: None, netmask: 255.255.255.0 }
1970-01-01, 00:00:04.777 [34m[INFO][0m [3m[90m usb: [0m Using UDC: "3f980000.usb"
1970-01-01, 00:00:04.777 [34m[INFO][0m [3m[90m usb: [0m 🔌 Initializing USB Manager
1970-01-01, 00:00:04.780 [34m[INFO][0m [3m[90m proxy: [0m 🛰️ Starting TCP server...
1970-01-01, 00:00:04.780 [34m[INFO][0m [3m[90m proxy: [0m 🛰️ TCP server bound to: [4m0.0.0.0:5288[24m
1970-01-01, 00:00:04.781 [34m[INFO][0m [3m[90m proxy: [0m 💤 waiting for bluetooth handshake...
1970-01-01, 00:00:04.784 [34m[INFO][0m [3m[90m usb: [0m 🔌 USB Manager: Disabled all USB gadgets
1970-01-01, 00:00:04.789 [34m[INFO][0m [3m[90m bluetooth: [0m 🥏 Bluetooth alias: [1m[32mWirelessAADongle-1c36a4[0m
1970-01-01, 00:00:04.836 [34m[INFO][0m [3m[90m bluetooth: [0m 🥏 Opened bluetooth adapter [1mhci0[0m with address [1mB8:27:EB:B6:9C:0E[22m
1970-01-01, 00:00:04.839 [34m[INFO][0m [3m[90m bluetooth: [0m 📣 BLE Advertisement started
1970-01-01, 00:00:04.909 [34m[INFO][0m [3m[90m bluetooth: [0m 📱 AA Wireless Profile: registered
1970-01-01, 00:00:04.912 [34m[INFO][0m [3m[90m bluetooth: [0m 🎧 Headset Profile (HSP): registered
1970-01-01, 00:00:04.912 [34m[INFO][0m [3m[90m bluetooth: [0m ⏳ Waiting for phone to connect via bluetooth...
1970-01-01, 00:00:04.913 [34m[INFO][0m [3m[90m bluetooth: [0m 🥏 Enumerating known bluetooth devices...
1970-01-01, 00:00:04.917 [34m[INFO][0m [3m[90m bluetooth: [0m 🧲 Trying to connect to: 7C:F0:E5:07:1B:92 ([1m[34mOnePlus Nord 4[0m)
1970-01-01, 00:00:06.906 [34m[INFO][0m [3m[90m bluetooth: [0m 🎧 Headset Profile (HSP): connect from: [1m7C:F0:E5:07:1B:92[0m
1970-01-01, 00:00:06.907 [34m[INFO][0m [3m[90m bluetooth: [0m 🔗 Device 7C:F0:E5:07:1B:92 ([1m[34mOnePlus Nord 4[0m) connected
1970-01-01, 00:00:08.503 [34m[INFO][0m [3m[90m bluetooth: [0m 📱 AA Wireless Profile: connect from: [1m7C:F0:E5:07:1B:92[0m
1970-01-01, 00:00:08.503 [34m[INFO][0m [3m[90m bluetooth: [0m 📲 Sending parameters via bluetooth to phone...
1970-01-01, 00:00:08.503 [34m[INFO][0m [3m[90m bluetooth: [0m 🛜 Sending Host IP Address: 10.0.0.1
1970-01-01, 00:00:08.503 [34m[INFO][0m [3m[90m bluetooth: [0m 📨 stage #1 of 5: Sending [33mWifiStartRequest[0m frame to phone...
1970-01-01, 00:00:08.625 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: received 4 bytes: [00, 00, 00, 02]
1970-01-01, 00:00:08.625 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: MessageID = 2, len = 0
1970-01-01, 00:00:08.625 [34m[INFO][0m [3m[90m bluetooth: [0m 📨 stage #2 of 5: Received [33mWifiInfoRequest[0m frame from phone (⏱️ 121 ms)
1970-01-01, 00:00:08.625 [34m[INFO][0m [3m[90m bluetooth: [0m 🛜 Sending Host SSID and Password: AAWirelessDongle, ConnectAAWirelessDongle
1970-01-01, 00:00:08.625 [34m[INFO][0m [3m[90m bluetooth: [0m 📨 stage #3 of 5: Sending [33mWifiInfoResponse[0m frame to phone...
1970-01-01, 00:00:08.626 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: received 4 bytes: [00, 02, 00, 07]
1970-01-01, 00:00:08.626 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: MessageID = 7, len = 2
1970-01-01, 00:00:08.626 [34m[INFO][0m [3m[90m bluetooth: [0m 📨 stage #4 of 5: Received [33mWifiStartResponse[0m frame from phone (⏱️ 0 ms)
1970-01-01, 00:00:08.626 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: remaining 2 bytes: [18, 00]
1970-01-01, 00:00:16.462 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: received 4 bytes: [00, 02, 00, 06]
1970-01-01, 00:00:16.463 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: MessageID = 6, len = 2
1970-01-01, 00:00:16.463 [34m[INFO][0m [3m[90m bluetooth: [0m 📨 stage #5 of 5: Received [33mWifiConnectStatus[0m frame from phone (⏱️ 7836 ms)
1970-01-01, 00:00:16.463 [36m[DEBUG][0m (4) aa_proxy_rs::bluetooth: remaining 2 bytes: [08, 00]
1970-01-01, 00:00:16.463 [34m[INFO][0m [3m[90m bluetooth: [0m 🚀 Bluetooth launch sequence completed
1970-01-01, 00:00:16.463 [34m[INFO][0m [3m[90m proxy: [0m 🛰️ TCP server: listening for phone connection...
1970-01-01, 00:00:16.464 [34m[INFO][0m [3m[90m proxy: [0m 📳 TCP server: new client connected: [1m10.0.0.15:39988[22m
1970-01-01, 00:00:16.464 [34m[INFO][0m [3m[90m proxy: [0m 📂 Opening USB accessory device: [4m/dev/usb_accessory[24m
1970-01-01, 00:00:16.464 [34m[INFO][0m [3m[90m proxy: [0m ♾️ Starting to proxy data between TCP and USB...
1970-01-01, 00:00:16.464 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: before read
1970-01-01, 00:00:16.464 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: before read
1970-01-01, 00:00:16.470 [34m[INFO][0m [3m[90m usb: [0m 🔌 USB Manager: Enabled default gadget
1970-01-01, 00:00:16.470 [34m[INFO][0m [3m[90m bluetooth: [0m 📣 Removing BLE advertisement
1970-01-01, 00:00:16.470 [34m[INFO][0m [3m[90m bluetooth: [0m 🥷 Unregistering default agent
1970-01-01, 00:00:16.470 [34m[INFO][0m [3m[90m bluetooth: [0m 📱 Removing AA profile
1970-01-01, 00:00:16.471 [34m[INFO][0m [3m[90m bluetooth: [0m 🎧 Removing HSP profile
1970-01-01, 00:00:16.641 [34m[INFO][0m [3m[90m bluetooth: [0m 💤 Bluetooth adapter powered off
1970-01-01, 00:00:17.303 [36m[DEBUG][0m (6) aa_proxy_rs::usb_gadget: got uevent: UEvent {
    action: Change,
    devpath: "/devices/virtual/misc/usb_accessory",
    subsystem: "misc",
    env: {
        "OF_NAME": "usb",
        "MAJOR": "10",
        "SUBSYSTEM": "misc",
        "MINOR": "124",
        "OF_FULLNAME": "/soc/usb@7e980000",
        "DEVPATH": "/devices/virtual/misc/usb_accessory",
        "USB_UDC_DRIVER": "default",
        "USB_UDC_NAME": "3f980000.usb",
        "OF_COMPATIBLE_0": "brcm,bcm2835-usb",
        "DRIVER": "configfs-gadget.default",
        "ACCESSORY": "START",
[... 291 lines cut ...]
1970-01-01, 00:00:18.125 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: after write, 8 bytes
1970-01-01, 00:00:18.125 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: before read
1970-01-01, 00:00:18.190 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: after read, 889 bytes
1970-01-01, 00:00:18.190 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: before write
1970-01-01, 00:00:18.190 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: after write, 889 bytes
1970-01-01, 00:00:18.191 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: before read
1970-01-01, 00:00:18.191 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: after read, 278 bytes
1970-01-01, 00:00:18.192 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: before write
1970-01-01, 00:00:18.192 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: after write, 278 bytes
1970-01-01, 00:00:18.192 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: before read
1970-01-01, 00:00:18.338 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: after read, 57 bytes
1970-01-01, 00:00:18.338 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: before write
1970-01-01, 00:00:18.339 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: after write, 57 bytes
1970-01-01, 00:00:18.339 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: before read
1970-01-01, 00:00:18.341 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: after read, 114 bytes
1970-01-01, 00:00:18.341 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: before write
1970-01-01, 00:00:18.341 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: after write, 114 bytes
1970-01-01, 00:00:18.342 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: before read
1970-01-01, 00:00:18.342 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: after read, 57 bytes
1970-01-01, 00:00:18.342 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: before write
1970-01-01, 00:00:18.342 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB: after write, 57 bytes
1970-01-01, 00:00:18.343 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: before read
1970-01-01, 00:00:18.343 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: TCP: after read, 57 bytes
1970-01-01, 00:00:18.343 [36m[DEBUG][0m (1) aa_proxy_rs::io_uring: USB:
```

rohanb1216, 2025-02-25:
Seems like the logs were cut off, attaching the full file below. Hopefully it should clear it up. It only works for a split second after the cmdline changes, not sure why. Without cmdline changes, it does not work at all, as it seems my headunit requires accessory mode.

[aars.log](https://github.com/user-attachments/files/18965170/aars.log)

rohanb1216, 2025-02-25:
1. Yep, working on the original, but I was facing the same intermittent issues that prompted you to create this version :)
2. Will try and report back.
3. Works great with a cable, zero issues there.
4. Tried forgetting the device, re-burning the sd card, haven't tried clearing androidauto cache. 
5. What exactly do you mean by trying it in the car? This was all being done on the headunit in the car, do you mean unpairing android auto from the headunit?

manio, 2025-02-25:
> This was all being done on the headunit in the car, do you mean unpairing android auto from the headunit?

Yep - I remember that someone had to unpair/clear phone profile from car headunit to make it work back again - I mean rather a bluetooth connection profile. Anyway - maybe this is not the case, especially if it is working for you with the upstream project...

rohanb1216, 2025-02-25:
Set the timeout to 300 seconds but the same thing happened, I'll try a build with the keepalive implemented to see if that fixes it

manio, 2025-02-26:
I created [nodelay](https://github.com/manio/aa-proxy-rs/tree/nodelay) branch - can you try it?

rohanb1216, 2025-02-26:
Sure, let me compile and I'll try in a bit. I was also trying to see if there were any differences in connection approach between this and the upstream, and it seemed to be the same, will also go back to upstream to see if I can pull any useful logs there too.

manio, 2025-02-26:
Great, thanks!

rohanb1216, 2025-02-26:
getting a segfault when trying to run the compiled binary, compiling with arch on wsl. any config changes to make when building locally? or should i try docker build instead?

manio, 2025-02-26:
Yes, docker is good option if you have problems but it is building from `main` branch so you need to adjust the Dockerfile to switch to the branch like in the following pseudo-diff:
```diff
  RUN git clone https://github.com/manio/aa-proxy-rs .
+ RUN git checkout nodelay
  RUN cargo build --release
```

Let me know if you have problems, so I'll just build it for you (I don't want this change go to the main yet before I also test it more thoroughly)...

rohanb1216, 2025-02-26:
Was able to run the binary compiled by docker, but I was running into the same issue where the server starts and the image appears for a second, then nothing. It seems like the headunit can still detect that there's a device connected, but maybe there's no AA packets being sent? I've attached logs again to see why it may be happening.

[aars.log](https://github.com/user-attachments/files/18980675/aars.log)

manio, 2025-02-26:
Ok... so it seems this was blind shoot, anyway: it seems you attached the same log, is it possible?

manio, 2025-02-26:
This time it was 7.6sec before it stalled :(
To be honest no idea what could be the difference, especially because I was based on the Nisarg's code directly, keeping all timings/delays as he has worked out for his development period.
So it is working normally (in sense of time running) on the aawgd, right?

Are you maybe able to make a one-liner patch to his code and try it? I want to know if this is caused by his sophisticated packet analyzing - it is probably the only difference now...

rohanb1216, 2025-02-26:
It did work somewhat normally on aawgd, sometimes it would crash when starting a few times, then it would work normally.
Yep, more than happy to, just tell me what needs to be changed. I'll set up the build env in the meantime.

manio, 2025-02-26:
Oh great, it's always good to work with developer on the other side :)

Try to change this:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/package/aawg/src/proxyHandler.cpp#L77
to `false`
This will disable sending only "full frames" from phone to car. If it is the case then we will be thinking what to do next :)

manio, 2025-02-26:
Yeah, definitely it looks like you said (good news in terms of spotting the bug ;). So you have exactly same/similar behavior in the aawgd like in the aa-proxy-rs after using a patched version, right? Connecting for several seconds and stall?

rohanb1216, 2025-02-26:
Yep! Connection does not close fully, so it detects the phone, but android auto is not detected.

manio, 2025-02-26:
Rohan,
Much thanks for your help so far.
In this case I'll try to prepare a code for handling this problem ASAP.
stay tuned...

manio, 2025-02-26:
OK... I prepared something (in the same `nodelay` branch) but I didn't test it yet in my car. I'll try to do some test soon...
If you're brave enough and you're boring, then you can try it (but as I said it was not tested in a real environment so it may just not work)... or just wait until I test it :)
As you like...

rohanb1216, 2025-02-27:
Testing it on my commute today, lets see how it goes :)

manio, 2025-02-27:
Doesn't work also for me... trying to debug the problem...

manio, 2025-02-27:
Try the branch now... for the quick test it worked for me (sending the whole messages only)... I am now wondering if it helps to you. I was using it with a longer timeout for testing...

manio, 2025-02-27:
Hold on... I was testing the wrong binary :(

manio, 2025-02-27:
Rohan,
I've added several more commits. It is now properly reading the message sizes. I was testing it but my car radio seems doesn't like this method... It just stops after some time (4 seconds or so), but your car may be different (especially because the main branch is working for my car, while not for your). I think it's worth to try it...

manio, 2025-03-01:
@rohanb1216
How's it going?
I found the problem! The code is `nodelay` branch is now updated. It is working also for me, so I am really curious if it is finally working for you (sending the full message/frames code)...

rohanb1216, 2025-03-01:
I was just about to say, tried with the changes you mentioned before and AA was giving me a 'security check not passed' error. I'll recompile and try again then :)

rohanb1216, 2025-03-01:
It works! Connects in seconds and flawlessly! Thank you so much for putting in the effort to get it working, this will make my commute that much better :) my thanks

manio, 2025-03-01:
Oh... this is finally good news! Also thank you for your time testing this. My plan is to clean this code up and merge it to main branch... Maybe I'll ask you for test before I finally merge this...

Please also tell me if this is related with the legacy/`-l` switch? Maybe it is now working for you without the legacy mode? Can you check?

rohanb1216, 2025-03-01:
I'd be more than happy to test before you merge, and I'll report back on results without the legacy switch in a bit :)

rohanb1216, 2025-03-02:
Unfortunately seems like my headunit requires the legacy switch, but it does work flawlessly with it enabled.

manio, 2025-03-02:
OK, thanks for confirmation, so there are units which just need this. Could you please finally test the branch?
I've cleaned the code up. I am still thinking if I do this change unconditionally, under the legacy switch or under new switch...
Nevertheless please just make sure that it is still working as expected :)

rohanb1216, 2025-03-03:
Yep, just tested and it's working great :) If I was in your place, my decision would be based on whether or not this would be a breaking change for some users. So for max compatibility, under a new switch, unless I was able to test on a few devices that it worked for them too.

manio, 2025-03-03:
I'll think about it and test longer in my environment...
Thanks for the test!

iceteaSA, 2025-03-04:
@manio The nodelay branch fixed an issue where AA did not start up properly on first run when starting my car, previously it would always load, then when I plugged it out and in it would work. Now it works the first time. Not using the legacy method.

iceteaSA, 2025-03-04:
@manio Ok, while this does seem to improve the situation some, the issue happened again. 

Basically what happens is that the first time it connects, it seems to get stuck, which I suspect is because the head unit is not ready. I see my phone is connected to the Wi-Fi and I get a notification of Looking for Android Auto, but it just gets stuck here. plugging the pi out and in usually works.

And of course every time I take my laptop with to debug, it does not happen. 

Is there any way to persist/iterate log files?

manio, 2025-03-04:
> Is there any way to persist/iterate log files?

If you have this problem again... just connect with ssh to the Pi and grab the logs out of it...
Making it persistent is possible but you need some basic linux knowledge...

manio, 2025-03-08:
@rohanb1216
Merged into main, thank you for your help on this!
Can I only ask what car/headunit brand you have?

closed 2025-03-08 by manio

rohanb1216, 2025-03-08:
No worries! And I have a Sony XAV-AX5500 (aftermarket headunit)

rohanb1216, 2025-03-26:
I do get the occasional stutter, my main issue now is that there's disconnects that I need to reboot the pi to make it connect again.
What do you have in mind?

manio, 2025-03-26:
Damn, I am pretty sure that I've responded to your message and now I can see that it doesn't saved/posted :(
So one more time:
I also have similar problems as you (eg. random disconnections) and not stable working with full-frames mode enabled. I think I might know where the problem is. I am now working on some nice feature for this project, and when I'll finish it, I'll try to get back to this and try to optimize it...
Stay tuned for the updates.

manio, 2025-03-28:
Hi,
I didn't have opportunity to test it deeply (for longer period) but I've tested it for some time.
For my quick look it was working OK (at least no disconnection or music stuttering, while the old code disconnects for sure).

Here it is: [fullframes2 branch](https://github.com/manio/aa-proxy-rs/pull/new/fullframes2)
I'll be glad if you test it and compare to the main code you are using currently.

rohanb1216, 2025-03-28:
Will take a look, thanks again for all your work!

iceteaSA, 2025-03-30:
@manio I tried your fullframes2 branch with the`-m` flag, but I think I somehow messed up the certs. Could you possibly DM me the correct ones in a zip file?

rohanb1216, 2025-03-30:
I'm also having trouble with the certs, would appreciate some help obtaining them as well :)

manio, 2025-03-30:
Sure guys, drop me an e-mail so I'll reply you back with the certs. My e-mail is public in my profile.

manio, 2025-03-30:
One important thing - you should need a certs if you are using code from that branch!

rohanb1216, 2025-03-30:
Ah okay! I'll keep the certs handy in case then

rohanb1216, 2025-04-01:
Had my first opportunity to test it, was flawless for my entire drive to work :) I'll report back after a bit more driving but seems like it's good!

rohanb1216, 2025-04-07:
After several hours of driving/testing, I've faced zero disconnects and only a couple of stutters that I'd just chalk to wireless being wireless. Seems great!

manio, 2025-04-07:
Great! Thanks for your input and testing, @rohanb1216 :)
In this case I'll rework the full-frames to use this algorithm. It is also working better for me.

manio, 2025-04-17:
> In this case I'll rework the full-frames to use this algorithm. It is also working better for me.

Done in `main`

commit bafd7959 (2025-03-08): Merge branch 'full-frames'

Closes #18
commit daf51c63 (2025-04-17): use a mitm code for full frames / passthrough

Previous `full_frames` handled by *io_uring.rs* was not optimal because
it was reading only a single packet when processing input read data.
This was leading to stuttering/disconnections.

The MITM code is more optimal in this regard, so use it also for
`full_frames` feature for now.

Fixes: #18