aa-proxy/aa-proxy-rs #19: Modifying packets on the fly

Deadknight, 2025-02-26:
Is it possible to change the packets sent and received from android auto? I have an electric car and i want to expose some data through "androidx.car.hardware"

manio, 2025-02-26:
Hi,
The way that this project is working (and also the upstream Nisarg's project) is based on transparently transferring data between those two devices (phone and car) without changing any data in-the-middle.
The AndroidAuto protocol AFAIK is not even documented so it is even harder to do something with this stream, but even If we could I am rather against this because this could case additional problems...
What you want is exposing/using this API one one side - car or phone if you want to use it. Eventually run some additional soft on the Pi which connects eg. to the phone but using quite different data channel.
btw: I also have electric car ;)

For now I am closing this issue, feel free to reopen and/or ask if you have further questions.

closed 2025-02-26 by manio

Deadknight, 2025-02-26:
Thanks for your answer.

In my country connected services not working for my car, so car cannot calculate routes properly. As you know ev cars needs some connectivity. My goal is to activate "EV Routing for android auto" by spoofing maker model or other needed data in proxy to trick google maps to open this type of routing. Some manufacturers like Ford enabled this in their hu software.

The decryption and encryption code is available here:
https://github.com/tomasz-grobelny/AACS

Also there is a proto buf list of commands here:
https://github.com/Demon000/web-auto

All left is to combine these somehow. I don't know rust very well, is the source code of "aawgd" binary available?

Ev routing link:
https://blog.google/products/android/android-auto-new-features-ces24/

manio, 2025-02-26:
I am aware of those two projects you mentioned but I was not getting into its details.

The routing link you provided seems to me rather like a commercial note, not a technical documentation. Anyway - from my understanding, this is rather a thing/API which could be used to passing additional info from car to the phone/or the other way round. Pi in-between is currently only a bridge/gateway, so how would you like it to act as?
In other words - what is the real use case modifying packets (be a MITM) for this?
You want to enable EV features for google maps?

> is the source code of "aawgd" binary available?

Of course: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/tree/main/aa_wireless_dongle/package/aawg/src

Deadknight, 2025-02-26:
Yes my primary objective is to enable ev routing on google maps. (Which includes charge stops based on battery percentage). I know its a long shot but i will try.

It requires mitm packet change. Since Android Auto in car headunit (which i will refer as hu) only input output device, google maps on mobile device asks hu for car info. If i can intercept and change these packets (car info, battery level etc.) probably mobile device maps app will send ev interface to hu (i think).

gamelaster, 2025-02-26:
My proxy might be useful for this: https://github.com/gamelaster/opengal_proxy

manio, 2025-02-26:
@Deadknight
The functionality looks indeed very cool. Something similar to what tesla's has out of the box :)

Deadknight, 2025-03-26:
i started, did lots of decompiling, used @gamelaster 's code for testing, added his code to aawgd. 

But a problem happened, whatever i send to gearhead it does not switch to ev routing mode.

I tried different hyundai and kia models which had ev_override setting in gearhead -> not worked
I tried sending fueltype electric and connector types -> not worked
I tried to simulate fuel sensors -> not worked

I also decompiled maps apk but its really a big project and hard to trace ev variables.

I currently think that maps has an geofencing situation for ev routing. Since i'm not at america i cannot prove this theory.

I also asked crankshaft developers here -> https://github.com/opencardev/crankshaft/discussions/730

They dont have anything on it too i think.

My goal is to add range and percent data reader to aawgd and feed it to FUEL sensor. If you have any more info about this lets contact, i can send you some more info privately. (Also there might be some solutions to some situations :D)