aa-proxy/aa-proxy-rs #20: WiFi disconnect on USB connection loss

jakobkordez, 2025-03-30:
A feature that would be quite useful in my opinion would be to either disable the WiFi hostapd or just disconnect all of the clients when the USB connection to the car is lost.
This is useful for the cars that still provide power to the USB port for a few minutes after the car has been turned off.

manio, 2025-03-30:
Hi,
I am not communicating directly with hostapd in any way. I think this rather belongs to the upstream project, and it was discussed there, eg:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/212
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/243
I am powering the Pi from the main HU USB port, so it is available only for 3 minutes or so, but also during charging (EV vehicle).

Can you provide the logs after the USB shutdown?
I am wondering if I am catching this power off in aa-proxy-rs...

manio, 2026-05-16:
Closing due to inactivity. Please reopen if you need.