aa-proxy/aa-proxy-rs #21: Special Features

Ioniq3, 2025-03-31:
Thanks a lot for the MITM; I think it will provide many new features, similar to those in the AAWireless dongle (https://www.aawireless.io)

I don't know how to implement it, but I have used the following ones a lot:

1) Activate developer mode to use Fermata Auto and Rimusic in Android Auto.
As Sirpo mentioned, it is done by "changing the name of the head unit (car) to 'Desktop Headunit.'"
https://xdaforums.com/t/aawireless-general-discussion-and-support.4247677/page-382#post-90001948

2) Disable the audio route from mobile to android auto
I used this when I let my children use my mobile in the car to watch YouTube. I didn't want the sound to play through the car speakers via Android Auto, but I still wanted to use Google Maps or Waze.

manio, 2025-03-31:
> Thanks a lot for the MITM; I think it will provide many new features, similar to those in the AAWireless dongle

Thanks and yeah, I hope that too :)

> As Sirpo mentioned, it is done by "changing the name of the head unit (car) to 'Desktop Headunit.'"

I think I know where to change that, and if I'm correct, this should be relatively easy. Can you test it if I create a special branch for you with this hardcoded name? But first I'd like to get your debug logs - to be sure that I am changing it the same way as in my car...

> Disable TTS sink (to avoid send play / stop command from mobile to android auto).

Unfortunately I cannot tell you more without some technical details what has to be changed and where to make this working...

Ioniq3, 2025-03-31:
Hi manio, if you tell how to get and what logs do you need I can get them.

Ioniq3, 2025-03-31:
> Thanks and yeah, I hope that too :)
>
>
> I think I know where to change that, and if I'm correct, this should be relatively easy. Can you test it if I create a special branch for you with this [...]

The feature is related to disable media sink, when I disable audio the youtube video sounds in the mobile phone not in the android auto car system, but if I try to play a song from a secound usb dongle it stop the youtube in the mobile pohone and if after this you touch the play icon in the youtube it stop the song in the car systems, so the feature disable the play / stop command between mobile phone and android auto.

manio, 2025-04-01:
This seems to me that it is related with `AudioFocusRequestNotification` frames. @gamelaster told me that he's got some idea for this :)

To obtain the logs: first of all you need recent version installed with `mitm` feature. This version build should [be here](https://github.com/manio/aa-proxy-rs/actions/runs/14169171682/artifacts/2850162354).
SSH in, then add `-m -d` to the startup parameter list and run it once in your car.
The logs should be in `/var/log/aa-proxy-rs.log`. You can even cat it on screen and save a ssh session to a file if you prefer this over scp-ing the file out of Raspberry.
Look [here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#getting-logs).

You also need keys and certs in `/etc/aa-proxy-rs` dir. Drop me an e-mail if you cannot follow the readme's instruction how to obtain it.

Ioniq3, 2025-04-01:
the logs, i get the ssl from https://github.com/borconi/headunit/blob/master/jni/hu_ssl.h#L29 , and use the hu and md the same (i think it is ok), but rpi3a doesn't connect with car:

1970-01-01, 00:18:28.082 [INFO] 🛸 aa-proxy-rs is starting, build: 20250331_101201, git: 20250331-948d31e
1970-01-01, 00:18:28.083 [INFO]  main:  📜 Log file path: /var/log/aa-proxy-rs.log
1970-01-01, 00:18:28.083 [INFO]  main:  ⚙️ Showing transfer statistics: 1m
1970-01-01, 00:18:28.086 [INFO]  proxy:  🛰️ Starting TCP server...
1970-01-01, 00:18:28.086 [DEBUG] (4) aa_proxy_rs: Found interface: Interface { name: "wlan0", flags: 69699, mac: [184, 39, 235, 124, 198, 39], address: 10.0.0.1, scope_id: None, netmask: 255.255.255.0 }
1970-01-01, 00:18:28.087 [INFO]  proxy:  🛰️ TCP server bound to: 0.0.0.0:5288
1970-01-01, 00:18:28.087 [INFO]  proxy:  💤 waiting for bluetooth handshake...
1970-01-01, 00:18:28.088 [INFO]  usb:  Using UDC: "3f980000.usb"
1970-01-01, 00:18:28.088 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-01, 00:18:28.090 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
1970-01-01, 00:18:28.091 [INFO]  bluetooth:  🥏 Bluetooth alias: WirelessAADongle-299372
1970-01-01, 00:18:28.102 [INFO]  bluetooth:  🥏 Opened bluetooth adapter hci0 with address B8:27:EB:83:39:D8
1970-01-01, 00:18:28.255 [INFO]  bluetooth:  📱 AA Wireless Profile: registered
1970-01-01, 00:18:28.258 [INFO]  bluetooth:  🎧 Headset Profile (HSP): registered
1970-01-01, 00:18:28.258 [INFO]  bluetooth:  ⏳ Waiting for phone to connect via bluetooth...
1970-01-01, 00:18:28.259 [INFO]  bluetooth:  🥏 Enumerating known bluetooth devices...
1970-01-01, 00:18:28.263 [INFO]  bluetooth:  🧲 Trying to connect to: 94:45:60:6F:F5:6B (Pixel 8)
1970-01-01, 00:18:29.046 [INFO]  bluetooth:  🎧 Headset Profile (HSP): connect from: 94:45:60:6F:F5:6B
1970-01-01, 00:18:29.048 [INFO]  bluetooth:  🔗 Device 94:45:60:6F:F5:6B (Pixel 8) connected
1970-01-01, 00:18:29.327 [INFO]  bluetooth:  📱 AA Wireless Profile: connect from: 94:45:60:6F:F5:6B
1970-01-01, 00:18:29.328 [INFO]  bluetooth:  📲 Sending parameters via bluetooth to phone...
1970-01-01, 00:18:29.328 [INFO]  bluetooth:  🛜 Sending Host IP Address: 10.0.0.1
1970-01-01, 00:18:29.329 [INFO]  bluetooth:  📨 stage #1 of 5: Sending WifiStartRequest frame to phone...
1970-01-01, 00:18:29.344 [DEBUG] (2) aa_proxy_rs::bluetooth: received 4 bytes: [00, 00, 00, 02]
1970-01-01, 00:18:29.345 [DEBUG] (2) aa_proxy_rs::bluetooth: MessageID = 2, len = 0
1970-01-01, 00:18:29.345 [INFO]  bluetooth:  📨 stage #2 of 5: Received WifiInfoRequest frame from phone (⏱️ 16 ms)
1970-01-01, 00:18:29.346 [INFO]  bluetooth:  🛜 Sending Host SSID and Password: AAWirelessDongle, ConnectAAWirelessDongle
1970-01-01, 00:18:29.346 [INFO]  bluetooth:  📨 stage #3 of 5: Sending WifiInfoResponse frame to phone...
1970-01-01, 00:18:29.357 [DEBUG] (2) aa_proxy_rs::bluetooth: received 4 bytes: [00, 02, 00, 07]
1970-01-01, 00:18:29.357 [DEBUG] (2) aa_proxy_rs::bluetooth: MessageID = 7, len = 2
1970-01-01, 00:18:29.358 [INFO]  bluetooth:  📨 stage #4 of 5: Received WifiStartResponse frame from phone (⏱️ 11 ms)
1970-01-01, 00:18:29.358 [DEBUG] (2) aa_proxy_rs::bluetooth: remaining 2 bytes: [18, 00]
1970-01-01, 00:18:30.118 [DEBUG] (2) aa_proxy_rs::bluetooth: received 4 bytes: [00, 02, 00, 06]
1970-01-01, 00:18:30.119 [DEBUG] (2) aa_proxy_rs::bluetooth: MessageID = 6, len = 2
1970-01-01, 00:18:30.119 [INFO]  bluetooth:  📨 stage #5 of 5: Received WifiConnectStatus frame from phone (⏱️ 760 ms)
1970-01-01, 00:18:30.119 [DEBUG] (2) aa_proxy_rs::bluetooth: remaining 2 bytes: [08, 00]
1970-01-01, 00:18:30.120 [INFO]  proxy:  🛰️ TCP server: listening for phone connection...
1970-01-01, 00:18:30.120 [INFO]  bluetooth:  🚀 Bluetooth launch sequence completed
1970-01-01, 00:18:30.124 [INFO]  usb:  🔌 USB Manager: Switched to accessory gadget
1970-01-01, 00:18:30.124 [INFO]  bluetooth:  🥷 Unregistering default agent
1970-01-01, 00:18:30.125 [INFO]  bluetooth:  📱 Removing AA profile
1970-01-01, 00:18:30.125 [INFO]  bluetooth:  🎧 Removing HSP profile
1970-01-01, 00:18:30.272 [INFO]  proxy:  📳 TCP server: new client connected: 10.0.0.15:52524
1970-01-01, 00:18:30.272 [INFO]  proxy:  📂 Opening USB accessory device: /dev/usb_accessory
1970-01-01, 00:18:30.273 [INFO]  proxy:  ♾️ Starting to proxy data between TCP and USB...
1970-01-01, 00:18:30.311 [INFO]  bluetooth:  💤 Bluetooth adapter powered off

thread 'main' panicked at src/mitm.rs:406:49:
not yet implemented
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

thread 'main' panicked at src/mitm.rs:432:48:
not yet implemented
1970-01-01, 00:18:45.275 [ERROR]  proxy:  🔴 Connection error: handling failed
1970-01-01, 00:18:45.276 [INFO]  proxy:  ⌛ session time: 15s 2ms 705us 515ns
1970-01-01, 00:18:45.276 [INFO]  main:  📵 TCP/USB connection closed or not started, trying again...
1970-01-01, 00:18:45.276 [INFO]  proxy:  💤 waiting for bluetooth handshake...
1970-01-01, 00:18:46.276 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-01, 00:18:46.277 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
1970-01-01, 00:18:46.278 [INFO]  bluetooth:  🥏 Bluetooth alias: WirelessAADongle-299372
1970-01-01, 00:18:46.289 [INFO]  bluetooth:  🥏 Opened bluetooth adapter hci0 with address B8:27:EB:83:39:D8
1970-01-01, 00:18:46.448 [INFO]  bluetooth:  📱 AA Wireless Profile: registered
1970-01-01, 00:18:46.451 [INFO]  bluetooth:  🎧 Headset Profile (HSP): registered
1970-01-01, 00:18:46.452 [INFO]  bluetooth:  ⏳ Waiting for phone to connect via bluetooth...

manio, 2025-04-01:
Look in your gmail mailbox.

gamelaster, 2025-04-01:
@Ioniq3  (just to note, you **cannot** use same cert for hu and phone)

Ioniq3, 2025-04-01:
Another try and it works, but I don't have the logs from the car (it is in another building,as soon i get them , it will post them), and It works with the same cert for hu and md.

Ioniq3, 2025-04-01:
[aa-proxy-rs.log](https://github.com/user-attachments/files/19552264/aa-proxy-rs.log)

manio, 2025-04-01:
Ok. Looks good. I will now change the name to the "Desktop Headunit", are you able to build and test a specific branch?

manio, 2025-04-01:
OK, here you are: https://github.com/manio/aa-proxy-rs/tree/mitm-devel
If you have problems I can provide you just the binary...
You need to run it the same (with `dpi` parameter set, otherwise it will not work).

Ioniq3, 2025-04-01:
[aa-proxy-rs.log](https://github.com/user-attachments/files/19554766/aa-proxy-rs.log)

It doesn't work to show Fermata Auto and RImusic.
I build with this:

git clone --recurse-submodules https://github.com/manio/aa-proxy-rs.git
cd aa-proxy-rs/
git checkout mitm-devel 
cargo build --release

copy aa-proxy-rs from aa-proxy-rs/target/arm-unknown-linux-gnueabihf/release
chmod a+x aa-proxy-rs
scp aa-proxy-rs root@10.0.0.1:/usr/bin

manio, 2025-04-01:
Yes, you built this correctly (I can see proper git revision in the log).
Now the question is: why it is not working :)
It seems that setting a make is not sufficient or I changed the wrong field.
You have an account there on xdaforums? Can you contact this guy who wrote it?

manio, 2025-04-01:
Can't find this string in his repo. Maybe he did it afterwards... I was also searching in all github repos and can't find something relevant...

manio, 2025-04-01:
Ok... I found something :)

manio, 2025-04-01:
Branch updated, pls give another try...

Ioniq3, 2025-04-01:
> Branch updated, pls give another try...

[aa-proxy-rs.log](https://github.com/user-attachments/files/19555745/aa-proxy-rs.log)

Great work.

Yes, now it works and it can use any apk:
https://github.com/AndreyPavlenko/Fermata
https://github.com/fast4x/RiMusic
https://github.com/KRTirtho/spotube

Hopeness there will be the same luck with the others features: media and tts sink

We can see a summary of the features aawireless provides https://www.reddit.com/media?url=https%3A%2F%2Fpreview.redd.it%2Faawireless-settings-for-kia-hyundai-genesis-v0-braceaqsutmb1.png%3Fauto%3Dwebp%26s%3D3ce6880e062c1937bc4341e31089794710d79b71:

remove tap restriction is to eliminate the 6 six touch restriction before show an message and waits for 10 seconds

manio, 2025-04-01:
Great and thanks for testing. I'll try with that sink when I have some time, will let you know...

manio, 2025-04-01:
Branch updated, pls give a try with that sink :)

manio, 2025-04-01:
Credits to @gamelaster :)
Thanks for testing.

Ioniq3, 2025-04-03:
For remove the audio focus, I mean, When disable media sink, if you are playing youtube on the mobile now it sound on it but if you want to play music on headunit it stops the youtube on mobile.

I tried to remove AUDIO_STREAM_GUIDANCE also and AA doesn't start

I remove packet of MESSAGE_AUDIO_FOCUS_REQUEST with pkt.payload.clear(); but it freezes, it don't know if it needs a first audio focus to mobile established or only message audio focus in one direction...

manio, 2025-04-03:
AUDIO_STREAM_GUIDANCE is rather a sink for navigation...

> I remove packet of MESSAGE_AUDIO_FOCUS_REQUEST with pkt.payload.clear()

I am not surprised, because you did it wrong :) You cannot just remove payload and leave the header :)

I'm not sure but maybe just ignoring the MESSAGE_AUDIO_FOCUS_REQUEST might help... no idea.
I'll try to do it and let you know when it's ready for test...

iceteaSA, 2025-04-04:
@manio Will the mitm mode possibly allow me to use the touch screen when in AA? My HU disable touch in AA, and have not found any way to enable it.

Ioniq3, 2025-04-04:
> @manio Will the mitm mode possibly allow me to use the touch screen when in AA? My HU disable touch in AA, and have not found any way to enable it.

I don't have any problem with mitm, i can use the touch screen of the aa , change app, use Google Maps input test and icons.

I think it is better upload logs for manio inspect.

manio, 2025-04-04:
> My car head unit software (Mazda Infotainment) disables the touch screen as soon as android auto launches, apparently it is a parameter sent to AA. I was wondering if with mitm it is possible to bypass this restriction, by having the pi not send the param through to my phone

It is really weird because: by default the MITM mode is doing just a forwarding (the same as you run in passthrough/default mode). In other words: it is passing the traffic unchanged. So when you connect your phone without MITM or directly via cable - you also have this screen touch disable? What is the sense of using AndroidAuto without touch screen? :)
I think we could do many things (like send/disable events), but as @iceteaSA said - your logs would be very useful.

manio, 2025-04-04:
> I'll try to do it and let you know when it's ready for test...

Done: https://github.com/manio/aa-proxy-rs/tree/mitm-devel2
It is just ignoring `MESSAGE_AUDIO_FOCUS_REQUEST` and not forwarding this. I don't know what effect it will does for you. You're beta tester, @Ioniq3 :)

Ioniq3, 2025-04-04:
> Done: https://github.com/manio/aa-proxy-rs/tree/mitm-devel2 It is just ignoring `MESSAGE_AUDIO_FOCUS_REQUEST` and not forwarding this. I don't know what effect it will does for you. You're beta [...]

It doesn't affect. 

The final feature it that the mobile ignore play/stop press in AA, for example i start radio or mp3 songs in HU and youtbe on mobile stop. In aawireless.io it is knows as "disable tts sink", maybe it could be the microphone stream as in wired earphone it could start/stop music with the button.

When disablle tts sink is conected the input for google assistant and the audio is in the micro ans speaker of the mobile, while in "disable media sink" the input and speaker of google assistant is all in the headunit.

iceteaSA, 2025-04-04:
@manio Sorry, I was unclear. This is not something caused by your software, but rather something my manufacturer baked into their implementation of AA, cause of reasons. And yeah, really brings down the usefulness of AA.

Don't ask me what they are, though. In the newer models of my car, this is apparently something you can toggle, which is why I think this might be something mitm can possibly solve for me. 

Which would be awesome, as I really hate the dial I have to use now. I'll try and get you those logs, but might be a few days. 

I rebuilt the image from the repo with docker, and for some reason Bluetooth didn't work after 😢 

Probably messed something up, so gonna redo it.

Do you want the debug logs with -m enabled?

manio, 2025-04-05:
> I tried to remove AUDIO_STREAM_GUIDANCE also and AA doesn't start

How you did it? Similarly as here: https://github.com/manio/aa-proxy-rs/blob/mitm-devel2/src/mitm.rs#L277 ?
It should be enough to change that name to AUDIO_STREAM_GUIDANCE and this should work - removing this service and the phone will not ask to open a channel for navigation audio - it would think that it is not available - just as we did with regular audio.

> Sorry, I was unclear. This is not something caused by your software, but rather something my manufacturer baked into their implementation of AA, cause of reasons. Don't ask me what they are, though. And yeah, really brings down the usefulness of AA.
> In the newer models of my car, this is apparently something you can toggle, which is why I think this might be something mitm can possibly solve for me.

Of course - I agree and I am trying to help - unfortunately I don't know how it should be done. I am also still learning the protocol and how things are working...

> I rebuilt the image from the repo with docker, and for some reason Bluetooth didn't work after 😢
> Probably messed something up, so gonna redo it.

I also think so - because the bluetooth code was not touched in any way recently. Maybe the phone is not connecting to the HU? I had this - the solution was to forget the devices in phone and do a clear new connection to the dongle/HU.

> Do you want the debug logs with -m enabled?

Yes, this might be helpful, especially - maybe you can see what frames are exchanged when you want play music or have that issue in your HU... Maybe you can also look at the Milek's doc you mentioned? Maybe you find something... at least some idea how to fix it...

To sum it up - I can try doing some changes using the mitm code, but I don't have knowledge how things should be done. If you figure it out and tell/propose changes - we could try to test various changes... :)

Ioniq3, 2025-04-05:
> How you did it? Similarly as here: https://github.com/manio/aa-proxy-rs/blob/mitm-devel2/src/mitm.rs#L277 ?
> It should be enough to change that name to AUDIO_STREAM_GUIDANCE and this should work - [...]

Thanks a lot @manio , i have little knowlodge about rust, how can i code to remove a service by the id that show me in the logs the service Discovery, to try one by one of them?

manio, 2025-04-05:
> how can i code to remove a service by the id that show me in the logs in the service Discovery message, to try one by one of them?

Using this:
```     msg.services.retain(|svc| svc.media_sink_service.audio_type() != AUDIO_STREAM_MEDIA);```
or basing on id:
```     msg.services.retain(|svc| svc.id() != 15);```

Those are retaining (leaving in services) - eg it leaves everything besides 15 - so effectively removing id=15

Ioniq3, 2025-04-05:
> Using this:
> ```     msg.services.retain(|svc| svc.media_sink_service.audio_type() != AUDIO_STREAM_MEDIA);```
> or basing on id:
> ```     msg.services.retain(|svc| svc.id() != 15);```
>
> Those are [...]

No luck.

Another try, keycode =85 is play/stop, how to remove from input_source_service?

manio, 2025-04-05:
> Another try @manio , keycode =85 is play/stop music, how to remove from input_source_service?

Try this (it should remove key 85 from the list):
```
                msg.services[2]
                    .input_source_service
                    .as_mut()
                    .unwrap()
                    .keycodes_supported.retain(|key|*key != 85);
```

iceteaSA, 2025-04-05:
@manio I was able to figure out the bluetooth problem, was caused by updates in [upstream](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle), when I rebuilt, I reflashed the image, and the updates made seems to have broken the Bluetooth. I just took your latest release image, and isntalled the build binary through scp, and all worked well. No idea why but it works again, so I am happy. 

Here are the logs with `-m` enabled:

[aa-proxy-rs.log](https://github.com/user-attachments/files/19615110/aa-proxy-rs.log)

Hopefully we can get through the touchscreen restriction Mazda has implemented, there are multiple forums that would sing your praises 🤣

naby-dev, 2025-04-05:
Hi!
I have this weird issue with AA in general and maybe this can help me fix things. 

Sometimes on my infotainment system, android will start up normally but there won't be any picture just a black screen. I believe AA is running in the background because I can hear the audio and whatnot but there's just a black screen.
I have no idea how to debug this and it seems to be rather random 

I was wondering if the MITM can improve things but restarting it if there's no pictures. I'm not sure if this is possible just thought I'd ask. 

I have a Samsung S22+ and my infotainment system is also Mazda. (Mazda3 2015)

iceteaSA, 2025-04-05:
@naby-dev Yeah, that happens to me often as well, USB or wireless. If you got mitm running with debug when it happens, the logs might show something. The Mazda implementation of AA is pretty bad, isn't it?

naby-dev, 2025-04-05:
> [@naby-dev](https://github.com/naby-dev) Yeah, that happens to me often as well, USB or wireless. If you got mitm running with debug when it happens, the logs might show something. The Mazda implementation of AA is pretty bad, isn't it?

It truly is terrible, they really half-assed things I know they use the excuse that they disable touch inputs for safety reasons but I wouldn't be surprised that they couldn't get it working. 

I'm glad I'm not the only one experiencing the issue. I will definitely need to give this a shot. I'll see if I can find the the steps to enable mitm

manio, 2025-04-05:
Do you guys have a Discord?
I think it would be easier if we move there instead of flooding this issue...
I just created a server for this project.
If you do, please head to: https://discord.gg/WAFNpNGn

Ioniq3, 2025-04-05:
Hi @manio .

Finally I get it.
The AA doesn't support remove the two STREAMS, AUDIO_STREAM_SYSTEM_AUDIO and AUDIO_STREAM_GUIDANCE 
so the solution is change the AUDIO_STREAM_GUIDANCE  for AUDIO_STREAM_SYSTEM_AUDIO    AudioStreamType when you want "disable tts sink" feauture of aawireless.io

So, This code works for me:

```
use crate::mitm::AudioStreamType::AUDIO_STREAM_SYSTEM_AUDIO;

....

            // disable tts sink
            if disable_tts_sink {
                msg.services[4]
                .media_sink_service
                .as_mut()
                .unwrap()
                .set_audio_type(AUDIO_STREAM_SYSTEM_AUDIO);
                info!(
                    "{} <yellow>{:?}</>: media tts disabled",
                    get_name(proxy_type),
                    control.unwrap(),
                );
            }
```

service[4] is from my car MESSAGE_SERVICE_DISCOVERY_RESPONSE:
media_sink_service {
    available_type: MEDIA_CODEC_AUDIO_PCM
    audio_type: AUDIO_STREAM_GUIDANCE
    audio_configs {
      sampling_rate: 16000
      number_of_bits: 16
      number_of_channels: 1
    } so the best solution is detect the id of the service AUDIO_STREAM_GUIDANCE before change it but I don't know enough rust

Another feauture of "Remove tap restriction" I think is removing:
      sensor_type: SENSOR_SPEED
     because it is said as disclaimer "could affect navigation apps :) "

manio, 2025-04-05:
@Ioniq3 I am impressed, good job! :)

Ioniq3, 2025-04-05:
> [@Ioniq3](https://github.com/Ioniq3) I am impressed, good job! :)

@manio 

It is important that disable tts goes before media link because it can change the order of the id maybe:

            // disable tts sink
            if disable_tts_sink {
                msg.services[4]
                .media_sink_service
                .as_mut()
                .unwrap()
                .set_audio_type(AUDIO_STREAM_SYSTEM_AUDIO);
                info!(
                    "{} <yellow>{:?}</>: media tts disabled",
                    get_name(proxy_type),
                    control.unwrap(),
                );
            }
            
            // disable media sink
            if disable_media_sink {
                msg.services
                    .retain(|svc| svc.media_sink_service.audio_type() != AUDIO_STREAM_MEDIA);
                info!(
                    "{} <yellow>{:?}</>: media sink disabled",
                    get_name(proxy_type),
                    control.unwrap(),
                );
            }

commit 5f6db810 (2025-04-06): mitm: Disable TTS sink

credits: @Ioniq3
closes: #21
commit 5f65bf1d (2025-04-06): mitm: Disable TTS sink

credits: @Ioniq3
closes: #21
closed 2025-04-06 by manio