aa-proxy/aa-proxy-rs #22: Sony XAV-AX5000 "To use android auto, please stop your car and follow the instructions on your phone"

jebbgrenham, 2025-04-26:
Opening this on Manio build, as this is not the same behaviour as aawg so presumably it has something to do with the rust rebuild.

On any boot or any attempted connection to the Pi, the head unit displays ""To use android auto, please stop your car and follow the instructions on your phone" but there are no notifications on the phone. It remains there for a few seconds, and then loses USB connection. After some time, it will reconnect and display the message again. Then it will disconnect, and repeat this whole loop many times. I think on this attached log session, it just gave up eventually? I can't recall.

Two logs attached. One had slightly different behavior perhaps, but I can't recall now as this was hours ago.  Anyway, neither boot worked.

[10.0.0.1_20250426093636.txt](https://github.com/user-attachments/files/19925195/10.0.0.1_20250426093636.txt)
[10.0.0.1_20250426092846.txt](https://github.com/user-attachments/files/19925194/10.0.0.1_20250426092846.txt)

manio, 2025-04-27:
Hi,
Unfortunately the logs are totally useless because my `aa-proxy-rs` explicitly logs all information into one file (to have all needed info in one place without additional "garbage"). The log by default is saved into `/var/log/aa-proxy-rs.log`.

But based on my experience with your Sony (we had a case #18 where @rohanb1216 also have `Sony XAV-AX5500`) unit I can blindly tell you that you need to enable `legacy` mode and also `full-frames` - at least for him it was not working without those.

Login with ssh, edit `/etc/init.d/S93aa-proxy-rs` and add:
`--legacy --full-frames`
parameters to line 10 with `AA_PROXY_RS_ARGS`.

jebbgrenham, 2025-04-27:
Well that make a marked improvement! It might make more sense (assuming the Pi assumes the role of USB host?) to have the Pi use the VID and PID of the peripeheral (the head unit) to set those flags rather than having to modify the build or the FS before using it.

It can get connected now at least - I haven't had time to do a test longer than initial connection and have some media play and navigate for 30 seconds. I also wasn't able to get logs because once it's connected now, there's no SSH access, even after disabling bluetooth on the phone unfortunately.

Thanks for this! It's at least as good as the C++ version, we'll see in the long term if it's even more reliable. I just really hope I can get logs if it does start random disconnects again.

You've got a tremendously useful project here and I can't thank you enough for your efforts.  I would be interested in contributing (perhaps I'll work on using the VID/PID to set flags for this head unit at least), and I would love to contribute to working on running a full stack android OS on this if it's technically possible.

manio, 2025-04-27:
> Well that make a marked improvement!

Great!

> I also wasn't able to get logs because once it's connected now, there's no SSH access, even after disabling bluetooth on the phone unfortunately.

In current Nisarg's build you need to explicitly enable the SSH, but you surely know that, so it has to be some other problem... no idea what exactly because I don't have reports confirming SSH access problems.

> Thanks for this! It's at least as good as the C++ version, we'll see in the long term if it's even more reliable. I just really hope I can get logs if it does start random disconnects again.

Yes, please. Reliability problems was my initial impulse to start working on this project :)

> and I would love to contribute to working on running a full stack android OS on this if it's technically possible.

If I remember correctly, this was covered by other projects - look into last chapter (links) on the main project page/readme.

jebbgrenham, 2025-05-10:
It's been non useable; just haven't had time to post. I'm actually wondering if I have a phone software issue now.
Random disconnects, random no-connection-at-all, random "please stop the car and follow instructions on your phone", random "usb device not supported". No rhyme or reason to when/why these errors occur. I need to try with a different phone...

[10.0.0.1_20250501083033.txt](https://github.com/user-attachments/files/20140300/10.0.0.1_20250501083033.txt)

manio, 2025-05-10:
In the logs, the relevant is this:
```1970-01-01, 00:10:17.508 [ERROR]  proxy:  🔴 Connection error: unexpected transfer stall```
This is the cause of disconnection - so no transfer for some time period and the daemon disconnects...

I hope that you'll be able to find the real cause of the problems...