aa-proxy/aa-proxy-rs #24: Error connecting bluetooth

d--bisset, 2025-06-25:
Hi,

Running a Pi 3A. Phone is a pixel 7a, car is a Mazda3 mk3.

Running using mitm mode. 

Usually all is fine and connection is fairly prompt, but occasionally it just doesn't connect. Typically when I an not in a place to grab logs.

I've managed to grab logs once in this situation. See attached. Possible race condition timing out when connection is in progress and leaving bluetooth in a bad state? Just a guess.

I'm running latest from your main branch. Happy to help debug but time is pretty limited.

[aa-proxy-rs1.log](https://github.com/user-attachments/files/20910150/aa-proxy-rs1.log)

d--bisset, 2025-06-25:
Have set a long timeout. Seemed to work okay in a quick test, will see how that goes over the next few days.

manio, 2025-06-28:
This may be the same problem as in #25...

douset, 2025-06-28:
> This may be the same problem as in [#25](https://github.com/manio/aa-proxy-rs/issues/25)...

Certainly sounds that way. I copied the test binary from that issue across and will test out over the next few days