aa-proxy/aa-proxy-rs #26: Future feature idea: Add climate/radio controls to unsupported vehicles once enabled in AA

cr08, 2025-07-03:
Thought about this as an extension to the MITM work and the EV features.

Since we've had the leaks of potential future support for integrated climate controls and vehicle radio tuner controls in AA, this seems like it'd align right along with the EV features as far as adding support for features that the vehicle manufacturer may not ever update their vehicles/headunits to support. As a Ford owner with Sync 3 and following closely with the 3.4 updates there, I'm not getting hopes up on them adding that support in this late in its life cycle.

Obviously this will require a bit more reverse engineering on a per vehicle basis in addition to now needing to do bi-directional CAN communication instead of a read only method like the EV functionality but I imagine if these features are added, it'd be a similar situation: Provide the baseline AA-side interface and leave the CAN side more on users to do the leg work.

manio, 2025-07-03:
> Since we've had the leaks of potential future support for integrated climate controls and vehicle radio tuner controls in AA, this seems like it'd align right along with the EV features as far as adding support for features that the vehicle manufacturer may not ever update their vehicles/headunits to support. As a Ford owner with Sync 3 and following closely with the 3.4 updates there, I'm not getting hopes up on them adding that support this late in its life cycle.

you think this would work on regular wired/wireless AndroidAuto as another extension?
I know this is possible but with Android Automotive OS (AAOS) not AA currently.

> Obviously this will require a bit more reverse engineering on a per vehicle basis in addition to now needing to do bi-directional CAN communication instead of a read only method like the EV [...]

Yeah, you feel this correctly.

gamelaster, 2025-07-03:
About radio stuff in the proto, radio was planned feature at beginning of android (around 2015-2016???), it was supported for a short time, and then Google deprecated it. Although, they might start to support it again, who knows.

cr08, 2025-07-03:
> About radio stuff in the proto, radio was planned feature at beginning of android (around 2015-2016???), it was supported for a short time, and then Google deprecated it. Although, they might start to support it again, who knows.

Fair enough! Hopefully they decide to maintain the existing protocol specs if that's the case. As I understand it that's all been reverse engineered with no official documentation from Google.

gamelaster, 2025-07-03:
@cr08 the current proto we have is dumped from DHU 1.x, so it's almost the same from Google, we did not needed reversing. Sadly, newer additions to the proto is mostly reverse engineered, with some names and stuff leaked from various sources.

Deadknight, 2025-07-04:
@cr08 There is a new service (currently in development i think) from google that controls various car functions and info. If that thing goes live somehow, there might be a chance

manio, 2025-07-04:
> What's weird yet frustrating is I have yet to see any screenshots/pictures/confirmations/whatever of this functionality working either on a production vehicle or even being emulated.

If you're talking about HVAC it's here on the link you mention:
[image]