aa-proxy/aa-proxy-rs #28: Toyota Aygo (Pioneer) Bluetooth connects and disconnect

toffyjan, 2025-07-09:
I’ve tried different USB cables and flashed multiple versions, but the Bluetooth keeps connecting and disconnecting. It stays on 'Waiting for Android Auto'. I'm experiencing the same issue with the latest Wireless Android Auto Dongle. It is Toyota Aygo. Android auto works fine with usb cable. I appreciate your work. 

[image]
[image]

manio, 2025-07-09:
> I know you can't help much without logs, but maybe you've encountered this kind of issue before.

The problem is that this 'Waiting for Android Auto' is very common and doesn't specify the exact problem. So really can't tell you more without the logs. Your screenshots doesn't help either :(

> I'm able to connect to the AAWireless Wi-Fi by entering the password ConnectAAWirelessDongle, even though it shouldn't be necessary—it should all happen automatically after the Bluetooth connection.

That's good, so it would be easier for you to get the logs :)
Exactly as you said - directly connecting to WiFi is not necessary at all in the "normally working" process.
If you're not in a hurry I am planning to add option for fetching logs via webserver which I added recently...
Please also consider joining us on Discord so it will be more direct help :)

toffyjan, 2025-07-10:
> > Thank you for your time. I can wait for the option to fetch logs via the web server. In the meantime, I'm trying to gather some logs, but it's a bit of a headache in this case.
> 
> OK... here is the most recent version with webserver and `Download log` option: https://github.com/manio/aa-proxy-rs/actions/runs/16186975037/artifacts/3501467051
> 
> What you need to do:
> 
> 1. Flash the proper image to your raspberry
> 2. Try several times to connect to dongle bluetooth from your phone
> 3. If it is still not working, connect to the dongle WiFi (you need to set a password for this, but you're probably aware as you was doing it before)
> 4. Enter your PI address in the browser to access embedded webserver
> 
> Then you should have this blue button to fetch the logs: <img alt="Image" width="2000" height="1014" src="https://private-user-images.githubusercontent.com/583157/464528815-820d9ba6-9d61-47ee-ba25-4f747e07429c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIxMzExNDksIm5iZiI6MTc1MjEzMDg0OSwicGF0aCI6Ii81ODMxNTcvNDY0NTI4ODE1LTgyMGQ5YmE2LTlkNjEtNDdlZS1iYTI1LTRmNzQ3ZTA3NDI5Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxMFQwNzAwNDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zM2E5NDVjNGU1NDU2ZTMwYjU3OTk1Yjc3N2U1MWMxM2IyZWM3MjE0Y2E1ZDU2OWVjMjJmNjMxOGFlZmU4MmM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.iYVyltPU9fq1-6RnDNzBGU21gl6FtMuhJatKHA3AX9Q"> 5. Give me the logs so I could see why it is not working
> 
> > Could you please point me in the right direction regarding how to join the Discord.
> 
> Just click [here](https://discord.gg/c7JKdwHyZu).
> 
> > Another thing I'm considering is that maybe the USB port in the car doesn't provide enough power to run the Raspberry Pi 2 W (enough power need for raspberry pi 2W, Bluetooth and wifi)
> 
> I don't think so... Pi02W is the most power efficient device from the supported Pi's. I hope you have a proper cable with USB Data+power (there are sometimes power-only for charging, but not for data).

I am using this cable https://www.amazon.de/-/en/Android-Mobile-Charging-Samsung-Motorola/dp/B0D41NKZ9R

manio, 2025-07-10:
> I am using this cable https://www.amazon.de/-/en/Android-Mobile-Charging-Samsung-Motorola/dp/B0D41NKZ9R

Should be fine:
[image]

toffyjan, 2025-07-10:
I collected various logs and zipped them all. I hope they will be helpful.

[aa-proxy-rs-logs.zip](https://github.com/user-attachments/files/21170279/aa-proxy-rs-logs.zip)

toffyjan, 2025-07-11:
> Thanks for the logs... Indeed same problem as reported in Nisarg's project.
> 
> In the mean time: does it help if you clear the cache?

 I used the 'Clear All Data' option in Android Auto's developer settings, but it didn't help. I'm going to try with an older Samsung Galaxy running Android 11 and see what happens. I haven't cleared the cache in the Android Auto app yet — I might try that and see if it helps.