aa-proxy/aa-proxy-rs #29: Cannot log into web interface

way-lo, 2025-07-17:
I'm attempting to log into the web interface of a 0.7.0 install on a Pi Zero 2 W.

My process:
1. Connect to WirelessAADongle-xxxxxx
2. Android Auto initiates
3. Phone automatically connects to SSID AAWirelessDongle 
(There is another identical entry in the available SSIDs, but I do not think we need to connect to this one?  I also could not find the password anywhere.  The original nisargjhaveri project lists it as ConnectAAWirelessDongle, but that does not work)

I then use a web browser on the phone to try to access http://10.0.0.1.  No connection can be established.

Maybe I'm missing something simple?

way-lo, 2025-07-17:
I was able to make some appropriate edits but I am still encountering the same issue.

1. I found that the AAWirelessDongle wireless password was commented out in the main config file.  I enabled it and was able to log into it.
2. I edited the webserver address in aawgd.conf from 0.0.0.0:80 to 10.0.0.1:80.  Still not able to get the web server to return anything.

manio, 2025-07-17:
Hi,
I can see you figured out how to set a password for WiFi.

> I edited the webserver address in aawgd.conf from 0.0.0.0:80 to 10.0.0.1:80. Still not able to get the web server to return anything.

No, this is not error: 0.0.0.0:80 is a "bind address", 0.0.0.0 means that it will listen on all available interfaces on port 80 (standard web port). It's odd that it's not working - I tested it multiple times...
Normally I would ask you for aa-proxy-rs logs (there is a line about server start) but you can only retrieve it when enable SSH server...

Maybe I have to update manual and maybe you are not aware: due to Android restriction you can use this web interface *ONLY* when AndroidAuto is not used! I mean: you need to connect manually to WiFi (as you discovered) and then enter the IP in your browser.

way-lo, 2025-07-17:
Thank you for your guidance.  I really appreciate it.  I am fairly unfamiliar with local networking so I appreciate your help as I stumble along.

I will revert the 0.0.0.0 change and re-attempt this.  My phone may have been connected to the dongle while my laptop was logged into the wifi access point.

way-lo, 2025-07-18:
Thanks again.  I'm reading over the information on the log file provided in the config file.

To verify, after reading up on the basics of tmpfs memory, by default this is stored only in RAM.

I'm just nitpicking a sentence I found in the google search.  "Data stored in tmpfs is lost when the system is restarted or the mount is unmounted."  But I presume the data is also lost when the system is shut down?

manio, 2025-07-18:
> To verify, after reading up on the basics of tmpfs memory, by default the log is stored only in RAM.

Yes, and you can change it to `/persistent` as noted.

> But I presume the data is also lost when the system is shut down?

Yes it is (if you are still using /var/log)