aa-proxy/aa-proxy-rs #30: Use car's existing WiFi

aadnehovda, 2025-07-21:
Just started looking at adding a rpi to my motorhome's (elderly) Pioneer AVIC-F88DAB, but I already have a wifi router installed. Does this project require AP mode or could the PI be connected to the normal wifi instead?

manio, 2025-07-21:
This project is using RaspberryPi wireless chip to make an AP, then your phone is connecting to this after Bluetooth handshake complete. Don't know exactly what you meant so ask if you want to know more...

aadnehovda, 2025-07-21:
After bt handshake the phone connects to the AP, gets an IP address and then connects to a tcp server running on the pi, right? But if the pi was already connected to the same wifi as the phone, then maybe it could simply instruct the phone to connect to the tcp server immediately, without the AP part?

Nornode, 2025-08-16:
Hey, I got here during a possible new feature request... 

I just got here to say that you can indeed use you phones wifi to something else. I use it to create a hotspot for the cars entertainment system while it's also connected to AA.

manio, 2025-08-16:
> Hey, I got here during a possible new feature request...
> 
> I just got here to say that you can indeed use you phones wifi to something else. I use it to create a hotspot for the cars entertainment system while it's also connected to AA.

Heh... funny I just responded in your issue that this is not possible :)