aa-proxy/aa-proxy-rs #36: Problem using Raspberry Pi Zero 2 W

bowzee, 2025-08-04:
Hi,

I was never able to connect, using either nisargjhaveri latest release or aa-proxy-rs v0.7.0 with a Raspberry Pi Zero 2 W and my car, a Nissan Kicks 2024. I copied these logs using v0.7.0 but I do not know if they are enough:

[aa_2.txt](https://github.com/user-attachments/files/21583749/aa_2.txt)
[aa-proxy-rs.log](https://github.com/user-attachments/files/21583748/aa-proxy-rs.log)

This is the only thing I see, for about a second:

[image]

Am I doing anything wrong? Is it a power issue?

manio, 2025-08-04:
Hi
From your logs I can see that the main problem is rather with USB. Some thoughts:
1. Is your USB cable 100% with power and data pins (not a charging-only cable)
2. Do you have it connected to USB OTG port in your Pi02W ?
3. Can you try disable legacy mode

manio, 2025-08-26:
... also. For making sure that AA is working, set this field to blank:

[image]
And then reboot and connect manually...

manio, 2025-08-27:
@bowzee I've reuploaded v0.9.2 with wifi+bt changes reverted. Indeed there was a problem. If you wish please try again.