aa-proxy/aa-proxy-rs #39: Audio dropouts with some cars

kadimgh, 2025-08-11:
I'm experiencing frequent audio dropouts in my Mazda CX-5 (2021). The issue is intermittent, but on average happens 1-3 times per minute. AA wired works without dropouts in the same car

At the same time Mazda 3 (2018) works much better averaging 0-1 dropouts per 30 minutes

Hardware: Raspberry Pi 4
WiFi: 5GHz channel 36
SW: aa-proxy-rs 0.7.0

I've also tried other experimental builds (https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/217#issuecomment-3118161153) and radio configurations, but the issue seems to follow the car

kadimgh, 2025-08-11:
> I’m guessing you’re using passthrough mode (not MITM), right?

yes

> You didn’t mention your phone, but maybe try testing with a different brand or a newer device...

It's Pixel 3a XL, which is now ~6 years old and was a budget phone to start with. I don't have a different Android device to try at the moment. I understand the phone might be involved in the issue, but I just wanted to highlight that there is a setup that works without issues (with Mazda 3), so it is worth to have the issue logged here as a known issue

> You could try the latest version...

I'll try 0.8.0 and report if there are any improvements

manio, 2025-08-12:
@kadimgh 
Can you please try this image?
https://skyboo.net/temp/aa/joka90/rpi02w_sdcard.img
It's AAWirelessDongle based.

manio, 2025-08-12:
I built it only for RPi02W. Do you like to build it yourself or you like me to try?

manio, 2025-08-12:
I was also using docker, but to build it you also need to run this inside container before build:
`sudo apt install libelf-dev`

kadimgh, 2025-08-13:
> I was also using docker, but to build it you also need to run this inside container before build: `sudo apt install libelf-dev`

I've built the [main brunch of joka90 fork](https://github.com/joka90/WirelessAndroidAutoDongle/commit/5c3e700eb9665107ff34a65ea3b6d40c3968a3e7) and installed on my RPi4, but unfortunately I can reproduce the "audio dropout every 20 seconds" with it even more consistently compared to aa-proxy-rs 0.8.0

Here is the [SD card image](https://drive.google.com/file/d/1xjlGkh1Cvqh18zBggxWdOeLjNoDdAYli/view?usp=sharing) if anyone is interested

manio, 2025-08-13:
Yeah good idea if you have such possibility, but keep in mind that AAWireless is based on different hardware and software (version TWO is using 5GHz WiFi for sure). Thinking this way you could also try other raspberry with 5GHz band :)

kadimgh, 2025-08-15:
I've got a no-brand $30 dongle from Amazon, likely based on Allwinner v851 SoC + 5GHz radio and so far it worked without dropouts with the same phone and car that aa-proxy-rs + RPi4 were having issues with. I realize this is not very actionable, but just another data point

[image]

manio, 2025-09-15:
> One additional thing I've noticed is that dropouts happen at intervals divisible by 20 seconds

Look here: https://www.androidauthority.com/android-auto-stuttering-music-3597082/
"Everything worked fine until August 27, when their music started skipping every 20 seconds"
Sound familiar? :)

luisfcorreia, 2025-10-03:
yup, also got some stutters but not 20 seconds apart
OnePlus 8T
MG MG5 EV

rpi02w

manio, 2025-10-03:
> yup, also got some stutters but not 20 seconds apart

newest release?

lasanthax, 2025-10-08:
I have been using version 10 without any issues. However, I updated to the latest version 14 last week, and since then, I have encountered audio problems. The sound is not missing sections of words or music; rather, it resembles distortion. its intermittent and there seems no pattern.

manio, 2025-10-08:
Hmmm ... there was no big changes from 0.10 to 0.12... (0.11 was broken)

The thing is, wider channels aren’t always better when it comes to Android Auto.
It’s more about latency - so a wider channel isn’t necessarily an improvement (and it’s also more susceptible to interference).
That's why I am asking if something specific has improved it. I'd like to isolate this. Besides some settings was set by default, eg short preamble or 802.11n, so it's irrelevant to put it in config.

hewasajumperboy, 2025-10-08:
> Thanks! Are you maybe able to isolate which setting helped you the most?

I'll work through isolating which helped the most.  My guess is 802.11n, short preamble, and rts mode in combo.  My pi isn't using a 40mhz channel, so that's out already; that setting was leftover from testing if the stuttering was bandwidth related.

manio, 2025-10-15:
Thanks for your input. Yeah if those values are really helpful I can consider adding these to config...

kadimgh, 2025-10-16:
@manio unrelated data point: several AAW/CP no-name dongles that I've seen had a setting for audio latency in the range 300-1000ms. I'm not sure if it did anything, but possibly there is a room for some buffering to fight the dropouts

manio, 2025-10-17:
@lasanthax Thanks - good to know, but now I am probably even more interested if it's better with the dev build I mention above...

lasanthax, 2025-10-18:
I appreciate you including max_unacked on the settings page. I have downloaded the beta build.

Excuse me, but what settings should I adjust to see whether audio shuttering gets better?

manio, 2025-10-18:
> Excuse me, but what settings should I adjust to see whether audio shuttering gets better?

Try without any changes first, but then try the value in range 1-99

manio, 2025-10-18:
@lasanthax you're using MITM or passthrough mode? I just discovered that my optimization fix was only for MITM...

manio, 2025-10-18:
enable MITM (you need certs for this) or wait until I build new beta...

lasanthax, 2025-10-21:
@manio 
Good news, 2 discoveries while testing your last beat version. 

1- Ran out of SD cards and had to use used 15 years old 256Mb SD card (from old Blackberry) outperform new 8GB & 16GB Sd's (I have confirmed this by putting troubled v0.10 with shuttering on old SD card, observed very minimal shuttering without any confing changes, so all this possibly related to phone's OS and Pi's OS performance?)

2- Asuning max_unacked mean the gap between an account of all tranmitted pacets, I have change this value to =1, this have also resolved "Hay Google" voice prompt, now it work as expected, then have tested with = 10, this have worsen voice prompt and started to not detedct words in between. so reverted back to =1 now it works good. Shutter is intermittant/random but very minilal and bareable.

Hope this help, 
I'm keen on getting MITM,  set up now more than ever, is there any guide or instruction for Certs etc..