aa-proxy/aa-proxy-rs #41: Auto enable hotspot for the phone

Nornode, 2025-08-16:
Hi, 

Thought I'd start a conversation about my two "needs" on top of what that's project already does. 
Thank you so much for making the effort building and maintaining this! I really thick is awesome! 
Thanks! 

To my question:
In my Volvo V90 (pre google) I use my phone's hotspot to play Spotify on the entertainment system (better sound quality).
Can a connection to the AA system also trigger the phone to enable it's WiFi hotspot?
Ultimately this could possibly be a config entry somewhere.

I'll create another issue for the other wish I have...

manio, 2025-08-16:
> In my Volvo V90 (pre google) I use my phone's hotspot to play Spotify on the entertainment system (better sound quality).
Can a connection to the AA system also trigger the phone to enable it's WiFi hotspot?
Ultimately this could possibly be a config entry somewhere.

This is rather not possible - the WiFi device is currently used explicitly by AndroidAuto, so you probably cannot use it as a hotspot in the same time - and this is Google limitation. For similar reason we plan to use a bluetooth for communication with the RPi (normal network is not accessible)...

Nornode, 2025-08-16:
It actually is possible and I do this regularly with my Google Pixel 9 Pro.

I have tried to get other automations to toggle the hotspot capability (with Tasker app) on but fails due to restrictions on an Android API access level.