aa-proxy/aa-proxy-rs #45: Issue connecting - MITM and Pixel 7

vandsh, 2025-08-22:
Hi, 

Love the project and can't wait to get it working. 
Had tried before using WirelessAADongle and then with aa-proxy-rs, my experience is as follows; 

- go thru setup, click on BT wireless dongle 
- looks like it tries to pair, disconnects after a few seconds
- tries again 

It seems like it's iterating thru the device loop and not connecting completely. 

Additionally, when I enabled MITM, I saw an error about a missing pem cert.

[20250821180734_aa-proxy-rs.txt - with MITM](https://github.com/user-attachments/files/21929445/20250821180734_aa-proxy-rs.txt) 

[20250821182957_aa-proxy-rs.txt - no MITM](https://github.com/user-attachments/files/21929452/20250821182957_aa-proxy-rs.txt)

Current device is a Google Pixel 7
Head Unit is u connect 8.4
Raspberry pi zero 2 w

Any advice or direction is much appreciated. I'll see if I can get a hold of another device (non Pixel) to see if it's phone, image or raspberrypi related. If need any more details, let me know!

vandsh, 2025-08-22:
Ok sweet, thanks for that suggestion. I cleared it out and was able to connect, now I hardcoded it to my phones bluetooth MAC address and it seemed to connect. I will test it both without and with MITM but I think this may have done the ticket.

vandsh, 2025-08-25:
v0.9.1 seems to have settled whatever issue there was with the bluetooth connection; no longer need to hardcode/empty and manually select, but still seeing the odd `Error opening USB accessory: device or resource busy`

[aa-proxy-rs_1.log first log](https://github.com/user-attachments/files/21959532/aa-proxy-rs_1.log)
[aa-proxy-rs_hub4blt.log checked headunit before bluetooth](https://github.com/user-attachments/files/21959533/aa-proxy-rs_hub4blt.log)

manio, 2025-08-25:
Thanks... now: can you also please attach your /var/log/messages here?

vandsh, 2025-08-25:
Messages attached.

[messages.txt](https://github.com/user-attachments/files/21971386/messages.txt)

manio, 2025-08-25:
OK... I have a follow-up question because this looks a bit strange. At first glance, I'd suspect just some kind of USB issue, but tell me this:
Was this Raspberry Pi used for anything else before, and did the USB work properly then?
Is the same cable working when you connect your phone directly?

manio, 2025-08-25:
> When you say connecting to the phone directly, you're saying plugging the pi right into the phone?

I mean plug the phone to your HU using USB cable. Is it working then?