aa-proxy/aa-proxy-rs #49: Need help?

GhostRider5666, 2025-08-26:
Raspberry Pi Zero 2W ,I can't get it to work on my 2022 Corolla. Everything is connected but it just says looking for android auto. I'm on the latest build.

manio, 2025-08-26:
If it doesn't help download and give me logs...

GhostRider5666, 2025-08-26:
[20250826033119_aa-proxy-rs.log](https://github.com/user-attachments/files/21983373/20250826033119_aa-proxy-rs.log)

GhostRider5666, 2025-08-26:
Everything works now

GhostRider5666, 2025-08-26:
One thing that doesn't work for me anymore is connecting it to its WiFi for changing settings?

manio, 2025-08-26:
> One thing that doesn't work for me anymore is connecting it to its WiFi for changing settings?

Hmmm... it is working for me. Can you tell me how can I reproduce this?

GhostRider5666, 2025-08-28:
The cable I had was bad, If I touch the cable, the board would reboot. Fixed it with another cable.