aa-proxy/aa-proxy-rs #51: Good pi0w experience here too with v0.9.1 now

lasanthax, 2025-08-26:
I wanted to provide an update regarding the Pi0W performance. Initially, I suspected external interferences to be the primary issue, but after further investigation, it appears I was mistaken. I've shielded the Pi0W and relocated it away from the centre console. 

Unfortunately, the overall outcome has not improved significantly. However, I have noticed it does pick up interferences from nearby TRAM power lines and substations.

Previously, I was running version 0.8.0 with an overclocked configuration of 1050MHz, complemented by a substantial heatsink to mitigate audio issues. The audio quality at that time was rated 8/10.

Currently, I have installed version 0.9.1 and have been running it for a day without any overclocking. I’m pleased to report that there have been no significant issues, and the audio quality has markedly improved to 11/10.

**Stick on foil** 
[image]

[image]

**Aluminium foil shield** 

[image]

lasanthax, 2025-08-31:
Thank you for your continued efforts. 
I have tested version 0.9.2 and observed a slight jitter in the audio quality. I reverted to version 0.9.1 as a reference point for further comparison, upon which I conducted additional tests. After returning to version 0.9.2 for a while, I concluded that there were no significant differences in connection stability or audio performance between the two versions I could notice.

I would also appreciate any updates on getting the voice activation feature, specifically the "OK Google" command, operational at some stage down in the future.

manio, 2025-08-31:
Thank you for testing this.
Regarding "OK Google" - can you revert to a working version to confirm that it is working? I saw a reports regarding this problem but still I have not idea what might be causing this...

manio, 2025-09-01:
Thanks, I see but I was only asking you to confirm last working "OK Google" version (mine or Nisarg's)

lasanthax, 2025-09-01:
I apologise for any confusion. That feature has not functioned for me , whether using your version or Nisarg's. Until just ten minutes ago, 

I attempted to replicate the error before gathering logs for review. Suddenly, it began responding to my voice.

The only variable I can identify is that I have typically executed this command while driving, rather than in a quiet environment with the car off. Since then, I have been leaning closer to the microphone, and it appears to be working as expected.

I suspect the issue may be related to the microphone gain level in wireless mode. Is there a way to increase the microphone gain while connected wirelessly?

manio, 2025-09-01:
I have no idea :)

lasanthax, 2025-10-08:
> Just to add my two cents with v0.10.0 Audio very much depending on the phone I'm using. Samsung A12 - spotty at best A56 - pretty much works all the time
> 
> RPI-0W here too. Latest AA and play services at time of writing made no difference. Seems to be related to the phone (based on highly limited empirical testing & advice from the discord peeps)
> 
> I was testing stationary with engine and AC off. With he A12 if I pronounce more phonetically it strangely seems to work better. The A56 doesn't care, I can go full accent and it'll still mostly pick me up

Interesting find, yes!!!! my Samsung is a A32, but this may be a good reason to upgrade. ;-)