aa-proxy/aa-proxy-rs #53: fail to reconnect after disabling wifi

tomaszduda23, 2025-08-31:
reproduction steps:
1. connect AA
2. disable wifi on phone
3. trigger reconnect using bluetooth

recover
1. restart aa-proxy-rs

note
1. the same steps works fine with openDsh_dash 

```
aa-proxy-rs 
1970-01-05, 10:44:36.054 [INFO] 🛸 aa-proxy-rs is starting, build: 20250830_213310, git: -br#
1970-01-05, 10:44:36.055 [INFO]  main:  📟 SBC model: Xiaomi Mi A2 Lite
1970-01-05, 10:44:36.055 [INFO]  main:  ⚙️ config loaded from file: /etc/aa-proxy-rs/config.toml
1970-01-05, 10:44:36.055 [INFO]  main:  📜 Log file path: /var/log/aa-proxy-rs.log
1970-01-05, 10:44:36.058 [INFO]  main:  webserver running at http://0.0.0.0:80/
1970-01-05, 10:44:36.058 [INFO]  proxy:  🛰️ Starting TCP server for MD...
1970-01-05, 10:44:36.058 [INFO]  proxy:  🛰️ MD TCP server bound to: 0.0.0.0:5288
1970-01-05, 10:44:36.058 [INFO]  proxy:  💤 waiting for bluetooth handshake...
1970-01-05, 10:44:36.059 [INFO]  usb:  📬 Starting UEvent listener thread...
1970-01-05, 10:44:36.059 [INFO]  usb:  Using UDC: "7000000.usb"
1970-01-05, 10:44:36.059 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-05, 10:44:36.061 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
1970-01-05, 10:44:36.061 [INFO]  bluetooth:  🥏 Bluetooth alias: aa-proxy-730505
1970-01-05, 10:44:36.068 [INFO]  bluetooth:  🥏 Opened bluetooth adapter hci0 with address xx:xx:xx:xx:xx:xx
1970-01-05, 10:44:36.234 [INFO]  bluetooth:  📱 AA Wireless Profile: registered
1970-01-05, 10:44:36.239 [INFO]  bluetooth:  🎧 Headset Profile (HSP): registered
1970-01-05, 10:44:36.240 [INFO]  bluetooth:  ⏳ Waiting for phone to connect via bluetooth...

#First connection

1970-01-05, 10:44:41.275 [INFO]  bluetooth:  🎧 Headset Profile (HSP): connect from: yy:yy:yy:yy:yy:yy
1970-01-05, 10:44:41.558 [INFO]  bluetooth:  📱 AA Wireless Profile: connect from: yy:yy:yy:yy:yy:yy
1970-01-05, 10:44:41.559 [INFO]  bluetooth:  📲 Sending parameters via bluetooth to phone...
1970-01-05, 10:44:41.559 [INFO]  bluetooth:  🛜 Sending Host IP Address: 10.0.0.1
1970-01-05, 10:44:41.560 [INFO]  bluetooth:  📨 stage #1 of 5: Sending WifiStartRequest frame to phone...
1970-01-05, 10:44:41.609 [INFO]  bluetooth:  📨 stage #2 of 5: Received WifiInfoRequest frame from phone (⏱️ 48 ms)
1970-01-05, 10:44:41.609 [INFO]  bluetooth:  🛜 Sending Host SSID and Password: aa-proxy, aa-proxy
1970-01-05, 10:44:41.610 [INFO]  bluetooth:  📨 stage #3 of 5: Sending WifiInfoResponse frame to phone...
1970-01-05, 10:44:41.614 [INFO]  bluetooth:  📨 stage #4 of 5: Received WifiStartResponse frame from phone (⏱️ 4 ms)
1970-01-05, 10:44:42.318 [INFO]  bluetooth:  📨 stage #5 of 5: Received WifiConnectStatus frame from phone (⏱️ 702 ms)
1970-01-05, 10:44:42.318 [INFO]  bluetooth:  🚀 Bluetooth launch sequence completed
1970-01-05, 10:44:42.319 [INFO]  proxy:  🛰️ MD TCP server: listening for phone connection...
1970-01-05, 10:44:42.324 [INFO]  usb:  🔌 USB Manager: Enabled default gadget
1970-01-05, 10:44:42.324 [INFO]  bluetooth:  🥷 Unregistering default agent
1970-01-05, 10:44:42.325 [INFO]  bluetooth:  📱 Removing AA profile
1970-01-05, 10:44:42.325 [INFO]  bluetooth:  🎧 Removing HSP profile
1970-01-05, 10:44:42.360 [INFO]  bluetooth:  💤 Bluetooth adapter powered off
1970-01-05, 10:44:42.717 [INFO]  proxy:  📳 TCP server: new client connected: 10.0.0.19:46032
1970-01-05, 10:44:42.717 [INFO]  proxy:  📂 Opening USB accessory device: /dev/usb_accessory
1970-01-05, 10:44:42.717 [INFO]  proxy:  ♾️ Starting to proxy data between HU and MD...
1970-01-05, 10:44:42.718 [INFO]  proxy:  ⚙️ Showing transfer statistics: disabled
1970-01-05, 10:44:42.793 [INFO]  usb:  🔌 USB Manager: Received accessory start request
1970-01-05, 10:44:42.902 [INFO]  usb:  🔌 USB Manager: Switched to accessory gadget

#Wifi diconnected on phone

1970-01-05, 10:45:08.537 [ERROR]  proxy:  🔴 Connection error: read_input_data: timeout
1970-01-05, 10:45:08.538 [INFO]  proxy:  ⌛ session time: 25s 820ms 246us 605ns
1970-01-05, 10:45:08.538 [INFO]  proxy:  💤 waiting for bluetooth handshake...
1970-01-05, 10:45:08.538 [INFO]  main:  📵 TCP/USB connection closed or not started, trying again...
1970-01-05, 10:45:09.541 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-05, 10:45:09.546 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
1970-01-05, 10:45:09.547 [INFO]  bluetooth:  🥏 Bluetooth alias: aa-proxy-730505
1970-01-05, 10:45:09.557 [INFO]  bluetooth:  🥏 Opened bluetooth adapter hci0 with address xx:xx:xx:xx:xx:xx
1970-01-05, 10:45:09.722 [INFO]  bluetooth:  📱 AA Wireless Profile: registered
1970-01-05, 10:45:09.728 [INFO]  bluetooth:  🎧 Headset Profile (HSP): registered
1970-01-05, 10:45:09.728 [INFO]  bluetooth:  ⏳ Waiting for phone to connect via bluetooth...
[... 79 lines cut ...]
1970-01-05, 10:48:37.155 [INFO]  bluetooth:  📨 stage #2 of 5: Received WifiInfoRequest frame from phone (⏱️ 74 ms)
1970-01-05, 10:48:37.155 [INFO]  bluetooth:  🛜 Sending Host SSID and Password: aa-proxy, aa-proxy
1970-01-05, 10:48:37.155 [INFO]  bluetooth:  📨 stage #3 of 5: Sending WifiInfoResponse frame to phone...
1970-01-05, 10:48:37.166 [INFO]  bluetooth:  📨 stage #4 of 5: Received WifiStartResponse frame from phone (⏱️ 10 ms)
1970-01-05, 10:48:37.865 [INFO]  bluetooth:  📨 stage #5 of 5: Received WifiConnectStatus frame from phone (⏱️ 698 ms)
1970-01-05, 10:48:37.865 [INFO]  bluetooth:  🚀 Bluetooth launch sequence completed
1970-01-05, 10:48:37.866 [INFO]  proxy:  🛰️ MD TCP server: listening for phone connection...
1970-01-05, 10:48:37.871 [INFO]  usb:  🔌 USB Manager: Enabled default gadget
1970-01-05, 10:48:37.872 [INFO]  usb:  🔌 USB Manager: Received accessory start request
1970-01-05, 10:48:37.876 [INFO]  bluetooth:  🥷 Unregistering default agent
1970-01-05, 10:48:37.876 [INFO]  bluetooth:  📱 Removing AA profile
1970-01-05, 10:48:37.959 [INFO]  bluetooth:  💤 Bluetooth adapter powered off
1970-01-05, 10:48:37.982 [INFO]  usb:  🔌 USB Manager: Switched to accessory gadget

1970-01-05, 10:49:07.868 [ERROR]  proxy:  📵 TCP server: deadline has elapsed, restarting...
1970-01-05, 10:49:07.869 [INFO]  proxy:  💤 waiting for bluetooth handshake...
1970-01-05, 10:49:07.869 [INFO]  main:  📵 TCP/USB connection closed or not started, trying again...
1970-01-05, 10:49:08.871 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-05, 10:49:08.878 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
1970-01-05, 10:49:08.879 [INFO]  bluetooth:  🥏 Bluetooth alias: aa-proxy-730505
1970-01-05, 10:49:08.892 [INFO]  bluetooth:  🥏 Opened bluetooth adapter hci0 with address xx:xx:xx:xx:xx:xx
1970-01-05, 10:49:09.071 [INFO]  bluetooth:  📱 AA Wireless Profile: registered
1970-01-05, 10:49:09.077 [INFO]  bluetooth:  🎧 Headset Profile (HSP): registered
1970-01-05, 10:49:09.077 [INFO]  bluetooth:  ⏳ Waiting for phone to connect via bluetooth...
```

manio, 2025-09-04:
Please tell me... is there any difference when you enabled/disable `legacy` mode in config?

tomaszduda23, 2025-09-05:
If is disable legacy it doesn't connect at all.

manio, 2025-09-05:
Hmmm... I suspect that the errors may be related with the `f_accessory` module. Also `Timeout waiting for accessory start` means that device switching seems to not working... If you have a timeout, it means that the event listener is not obtaining any events: https://github.com/aa-proxy/aa-proxy-rs/blob/main/src/usb_gadget.rs#L25
Patch for this is also in buildroot. Btw without `legacy` you don't need this at all (that's it why I asked for this; eg my car HU doesn't need this and it is working flawless, same with DHU).

Can you check if you have properly propagated `/sys/kernel/config/usb_gadget` like this script is doing?
https://github.com/aa-proxy/buildroot/blob/main/external/rootfs_overlay/etc/S92usb_gadget.in
I can see that UDC was detected properly: "7000000.usb" so it should save this value when switching gadgets.

tomaszduda23, 2025-09-05:
> Can you check if you have properly propagated /sys/kernel/config/usb_gadget like this script is doing?

There are `/sys/kernel/config/usb_gadget/default/UDC` and `/sys/kernel/config/usb_gadget/accessory/UDC`. aa-proxy-rs toggle between them.

`/dev/usb_accessory` is never closed after first opening. Maybe aa tries to open this file twice? It uses io_uring which makes it hard to confirm using strace.

manio, 2025-09-05:
> `/dev/usb_accessory` is never closed after first opening. Maybe aa tries to open this file twice? It uses io_uring which makes it hard to confirm using strace.

Hmmm... it is opened here: https://github.com/aa-proxy/aa-proxy-rs/blob/main/src/io_uring.rs#L309 and should be dropped on next loop iteration, but maybe there is some problem... I need to check this again...

> BTW latest version does not work at all with my car. 0.9.2 was fine at least the first connection.

Are you maybe able to bisect or compare logs? Hard to tell why is this...

tomaszduda23, 2025-09-05:
I pulled wrong hash.
```
1970-01-10, 23:49:33.212 [ERROR]  proxy:  🔴 Connection error: unexpected transfer stall
thread 'main' panicked at /home/pmos/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.1/src/runtime/task/core.rs:378:22:
JoinHandle polled after completion
```

manio, 2025-09-06:
Hmmm... OK so this was a false lead... Thanks for testing... I'll try to figure this out...

manio, 2025-09-07:
Progress update:
I was trying to reproduce the problem with DHU...
I was diagnosing this with `lsof`...
(Un)fortunately it seems the `/dev/usb_accessory` descriptor is properly opened once and closed on next iteration so I don't see an fd leak there... But I'll try harder...

manio, 2025-09-26:
@tomaszduda23
Is it possible to build some test environment on Linux to reproduce this bug?

tomaszduda23, 2025-10-04:
There are 2 different bugs:
1. Android Auto does not work after disabling wifi and connecting again. This is the thing which I opened the issue for. I checked newer version on rpi0w and it still happens. After killing aa-proxy-rs process and opening it again it works once. The log:
```
1970-01-01, 00:01:26.310 [INFO] 🛸 aa-proxy-rs is starting, build: 20250926_095718, git: 3fb3a14-br#ffef988
1970-01-01, 00:01:26.313 [INFO]  main:  📟 host device: Raspberry Pi Zero W Rev 1.1
1970-01-01, 00:01:26.314 [INFO]  main:  ⚙️ config loaded from file: /etc/aa-proxy-rs/config.toml
1970-01-01, 00:01:26.317 [INFO]  main:  📜 Log file path: /var/log/aa-proxy-rs.log
1970-01-01, 00:01:26.323 [INFO]  proxy:  🛰️ Starting TCP server for MD...
1970-01-01, 00:01:26.328 [INFO]  proxy:  🛰️ MD TCP server bound to: 0.0.0.0:5288
1970-01-01, 00:01:26.330 [INFO]  proxy:  💤 waiting for bluetooth handshake...
1970-01-01, 00:01:26.332 [INFO]  main:  webserver running at http://0.0.0.0:80/
1970-01-01, 00:01:26.336 [INFO]  usb:  📬 Starting UEvent listener thread...
1970-01-01, 00:01:26.338 [INFO]  usb:  Using UDC: "20980000.usb"
1970-01-01, 00:01:26.339 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-01, 00:01:26.345 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
1970-01-01, 00:01:26.352 [INFO]  bluetooth:  🥏 Bluetooth alias: aa-proxy-530f2b
1970-01-01, 00:01:26.392 [INFO]  bluetooth:  🥏 Opened bluetooth adapter hci0 with address xx:xx:xx:xx:xx:xy
1970-01-01, 00:01:26.440 [INFO]  bluetooth:  📱 AA Wireless Profile: registered
1970-01-01, 00:01:26.454 [INFO]  bluetooth:  🎧 Headset Profile (HSP): registered
1970-01-01, 00:01:26.457 [INFO]  bluetooth:  ⏳ Waiting for phone to connect via bluetooth...
1970-01-01, 00:01:26.460 [INFO]  bluetooth:  🥏 Enumerating known bluetooth devices...
1970-01-01, 00:01:26.471 [INFO]  bluetooth:  🧲 Attempting to start an AndroidAuto session via bluetooth with the following devices, in this order: [xx:xx:xx:xx:xx:xx]
1970-01-01, 00:01:26.477 [INFO]  bluetooth:  🧲 Trying to connect to: xx:xx:xx:xx:xx:xx (PinkPhone)
1970-01-01, 00:01:28.157 [INFO]  bluetooth:  🎧 Headset Profile (HSP): connect from: xx:xx:xx:xx:xx:xx
1970-01-01, 00:01:28.163 [INFO]  bluetooth:  🔗 Successfully connected to device: xx:xx:xx:xx:xx:xx (PinkPhone)
1970-01-01, 00:01:28.925 [INFO]  bluetooth:  📱 AA Wireless Profile: connect from: xx:xx:xx:xx:xx:xx
1970-01-01, 00:01:28.927 [INFO]  bluetooth:  📲 Sending parameters via bluetooth to phone...
1970-01-01, 00:01:28.929 [INFO]  bluetooth:  🛜 Sending Host IP Address: 10.0.0.1
1970-01-01, 00:01:28.931 [INFO]  bluetooth:  📨 stage #1 of 5: Sending WifiStartRequest frame to phone...
1970-01-01, 00:01:28.988 [INFO]  bluetooth:  📨 stage #2 of 5: Received WifiInfoRequest frame from phone (⏱️ 53 ms)
1970-01-01, 00:01:28.990 [INFO]  bluetooth:  🛜 Sending Host SSID and Password: aa-proxy, aa-proxy
1970-01-01, 00:01:28.991 [INFO]  bluetooth:  📨 stage #3 of 5: Sending WifiInfoResponse frame to phone...
1970-01-01, 00:01:28.994 [INFO]  bluetooth:  📨 stage #4 of 5: Received WifiStartResponse frame from phone (⏱️ 0 ms)
1970-01-01, 00:01:29.700 [INFO]  bluetooth:  📨 stage #5 of 5: Received WifiConnectStatus frame from phone (⏱️ 703 ms)
1970-01-01, 00:01:29.702 [INFO]  proxy:  🛰️ MD TCP server: listening for phone connection...
1970-01-01, 00:01:29.704 [INFO]  bluetooth:  🚀 Bluetooth launch sequence completed
1970-01-01, 00:01:29.713 [INFO]  usb:  🔌 USB Manager: Enabled default gadget
1970-01-01, 00:01:29.713 [INFO]  bluetooth:  🥷 Unregistering default agent
1970-01-01, 00:01:29.714 [INFO]  bluetooth:  📱 Removing AA profile
1970-01-01, 00:01:29.714 [INFO]  bluetooth:  🎧 Removing HSP profile
1970-01-01, 00:01:30.172 [INFO]  usb:  🔌 USB Manager: Received accessory start request
1970-01-01, 00:01:30.293 [INFO]  usb:  🔌 USB Manager: Switched to accessory gadget
1970-01-01, 00:01:30.301 [INFO]  proxy:  📳 TCP server: new client connected: 10.0.0.19:33298
1970-01-01, 00:01:30.308 [INFO]  proxy:  📂 Opening USB accessory device: /dev/usb_accessory
1970-01-01, 00:01:30.311 [INFO]  proxy:  ♾️ Starting to proxy data between HU and MD...
1970-01-01, 00:01:30.402 [INFO]  proxy:  ⚙️ Showing transfer statistics: disabled
1970-01-01, 00:01:30.871 [INFO]  mitm/MD:  🔒 stage #1 of 3: SSL handshake: SSLv3/TLS write client hello
1970-01-01, 00:01:30.974 [INFO]  mitm/HU:  🔒 stage #1 of 2: SSL handshake: SSLv3/TLS write server done
1970-01-01, 00:01:31.092 [INFO]  mitm/MD:  🔒 stage #2 of 3: SSL handshake: SSLv3/TLS write finished
1970-01-01, 00:01:31.103 [INFO]  mitm/HU:  🔒 stage #2 of 2: SSL handshake: SSL negotiation finished successfully
1970-01-01, 00:01:31.105 [INFO]  mitm/HU:  🔒 SSL init complete, negotiated cipher: ECDHE-RSA-AES256-GCM-SHA384
1970-01-01, 00:01:31.200 [INFO]  mitm/MD:  🔒 stage #3 of 3: SSL handshake: SSL negotiation finished successfully
1970-01-01, 00:01:31.201 [INFO]  mitm/MD:  🔒 SSL init complete, negotiated cipher: ECDHE-RSA-AES128-GCM-SHA256
1970-01-01, 00:01:58.235 [ERROR]  proxy:  🔴 Connection error: read_input_data: EndpointIo timeout
1970-01-01, 00:01:58.236 [INFO]  proxy:  ⌛ session time: 27s 922ms 689us
1970-01-01, 00:01:58.237 [INFO]  main:  📵 TCP/USB connection closed or not started, trying again...
1970-01-01, 00:01:58.239 [INFO]  proxy:  💤 waiting for bluetooth handshake...
1970-01-01, 00:01:59.243 [INFO]  usb:  🔌 Initializing USB Manager
1970-01-01, 00:01:59.248 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
[... cut ...]
1970-01-01, 00:02:14.065 [INFO]  usb:  🔌 USB Manager: Switched to accessory gadget
```
2. https://github.com/aa-proxy/aa-proxy-rs/issues/53#issuecomment-3259691746 this happens only on musl libc. It seems that it is a bug in musl or https://github.com/tokio-rs/tokio-uring.

> Is it possible to build some test environment on Linux to reproduce this bug?

@manio I've checked already with openDsh_dash and it works fine. I can reproduce it 100% in the car though. If you give me new binary I can provide more logs or check other things.

manio, 2025-10-04:
Thanks for clarification...

> I've checked already with openDsh_dash and it works fine. I can reproduce it 100% in the car though. If you give me new binary I can provide more logs or check other things.

Hmmm... idealy If I could reproduce it myself... this way it should be easier to debug and solve this... Unfortunately I don't have Pi0W and I don't have this issue on Pi02W, Radxa and AAWireless2.
So regarding point 1: you just use Pi0W, start it, then disable WiFi in the phone and it cannot reconnect until you kill and restart aa-proxy-rs, right?

tomaszduda23, 2025-10-04:
> Unfortunately I don't have Pi0W and I don't have this issue on Pi02W, Radxa and AAWireless2.

pi0w does not matter. I checked another SBC and the same problem exists.

> So regarding point 1: you just use Pi0W, start it, then disable WiFi in the phone and it cannot reconnect until you kill and restart aa-proxy-rs, right?

Correct.

`1970-01-01, 00:01:31.201 [INFO]  mitm/MD:  🔒 SSL init complete, negotiated cipher: ECDHE-RSA-AES128-GCM-SHA256
` <- after this line AA works

`1970-01-01, 00:01:58.235 [ERROR]  proxy:  🔴 Connection error: read_input_data: EndpointIo timeout` <- this logs is printed after disabling wifi

after that it does not work anymore. Unless I kill aa-proxy-rs process.

manio, 2025-10-04:
Hmmm... so I need to investigate this further...
I was disabling WiFi on my phone million times and it always recovered perfectly fine from this...
The `EndopointIo timeout` is just a normal line telling that there is no data and it is restarting the connection.

tomaszduda23, 2025-10-04:
I checked a few times and there seems to be two variation of issue from point 1.
https://github.com/aa-proxy/aa-proxy-rs/issues/53#issuecomment-3368084250 if you look at this log:
`1970-01-01, 00:02:13.933 [INFO]  proxy:  🛰️ MD TCP server: listening for phone connection...` <- there is no connection request from the phone after that line. I would expect to see `1970-01-01, 00:01:30.301 [INFO]  proxy:  📳 TCP server: new client connected: 10.0.0.19:33298` but there is none.
If I do `nc 10.0.0.1 5288` from the laptop in the same time the log is printed.

manio, 2025-10-04:
Regarding 1a:
I’m having that issue too, though it’s pretty rare for me... This is definitely a phone problem... why it is not connecting? Who knows... In such case I generally need to make a "reset" AA sequence on my phone, that is:
1. disable wifi
2. disable bluetooth
3. discard all AA notifications
4. enable bluetooth

after this it generally connects successfully

you can also try to change to other phone for testing if this is occurring more often...

manio, 2025-10-04:
> Connection error: reader channel hung up

This is something I need to investigate further... it is looking like something has disconnected immediately...

manio, 2025-10-17:
Hold on... we've got a pmOS guy on Discord - I’ll try to ask him about it...

tomaszduda23, 2025-10-17:
To be clear. I also have the same problem on raspberry pi zero w.

tomaszduda23, 2025-10-17:
I dropped pmOS idea for a moment. The last logs which I shared where from pi0w

manio, 2026-05-16:
Closing this for now. The topic was also discussed further on Discord, and since quite a lot of time has passed already, a lot has probably changed in the codebase as well.