aa-proxy/aa-proxy-rs #55: RPi4 - Google Voice Commands missing words

teegles, 2025-08-31:
Latest build, everything else seems to work seamlessly however when doing voice commands it will just miss words or not work at all. I'm not sure which mic is used, but this wasn't an issue when hard wiring my phone in for AA.

Voice calls seem to work just fine, so it's doesn't appear to be a problem with a microphone hearing it.

RPi4 and Ford Fiesta ST with Sync 3 system

teegles, 2025-09-01:
Hey! I loaded up v0.9.1 and tried a few voice commands and it worked as expected. I'm not a developer by any means, but if you want me to test anything, lmk.

manio, 2025-09-01:
Thanks... So the problems started between 0.9.1 and 0.9.2?
https://github.com/aa-proxy/aa-proxy-rs/compare/v0.9.1...v0.9.2

manio, 2025-09-02:
> I'm not sure which version I was on before. Between this app and the other popular one, I have several card images I've put onto my microsd to tinker with my rpi4 and pi zero 2.
> 
> I know for sure when I opened the issue that I was on 0.9.2 and that going to 0.9.1 resolved it.

It's odd because between the two version I rather not suspect something special which could drop performance or something microphone related.
Look also here: https://github.com/aa-proxy/aa-proxy-rs/issues/51#issuecomment-3242027394

manio, 2025-09-09:
Look also here: https://github.com/aa-proxy/aa-proxy-rs/issues/51#issuecomment-3242027394