aa-proxy/aa-proxy-rs #57: 0.9.2 rarely connects successfully

kadimgh, 2025-09-03:
After upgrading to 0.9.2 I've noticed that it rarely connects successfully. On some rides it would not ever connect, while I'm seeing "USB device not recognized" message repeatedly on my HU. Sometimes it does connect though (sometimes from first and sometimes from one of the subsequent attempts). Subjectively 0.8.0 was able to connect almost every time

Logs (4 sessions total: power connection to enable the logs, two 10-minute rides with only one successful AA connection during the first ride, PC connection to collect the logs): 

[20250902172148_aa-proxy-rs.log](https://github.com/user-attachments/files/22105973/20250902172148_aa-proxy-rs.log)

manio, 2025-09-03:
Hi,
Looking at your logs it seems you have power problems. Your pi is rebooting in the middle of the logs constantly:
[image]
Also a lot of this kind:
[image]

My tips for you:
- Maybe it is also related with `legacy_mode` ? Can you try disable it?
- Check the quality of your USB cable - try better/thicker if you have...
- Try to disable `turbo_mode` in config in boot partition (comment out lines from 45 to 51 here: https://github.com/aa-proxy/buildroot/blob/main/external/board/rpi/config.txt#L45)
- Try to downclock your Pi, maybe try to change `pi5` to `pi4` here: https://github.com/aa-proxy/buildroot/blob/main/external/board/rpi/config.txt#L5
- What about your heatsink? You have one?
- 0.8.0 was pretty long ago - there was also 0.9.1 in the middle... can you also try this? Or if it doesn't work please go back to 0.8.0 and confirm that the problem is not there (and give me logs).

manio, 2025-09-04:
> BTW is there a way to enable persistent logs for SoC temperature via either aa-proxy logs or system logs?

not implemented but should be doable via some script, eg:

```bash
#!/bin/bash

# Path to the log file
LOG_FILE="/data/temp.log"

# Infinite logging loop
while true; do
    # Read temperature
    TEMP=$(cat /sys/class/thermal/thermal_zone0/temp)

    # Log with timestamp
    echo "$(date +'%Y-%m-%d %H:%M:%S') $TEMP" >> "$LOG_FILE"

    # Wait 10 seconds
    sleep 10
done
```

kadimgh, 2025-09-04:
After some more debugging it seems that the problem is still with my HU USB hub in one of the cars, not the temperature nor the cables. I can also see that the RPi4 red LED turns off in the problematic setup which should indicate the problem with power. I can still see 5.1-5.2V on the DMM and no dips during RPi4 operation, so I'm not sure what exactly is wrong with the power, but I'll try to boost power lines in the HU USB hub and see if it helps

kadimgh, 2025-09-10:
I've boosted the power lines in the hub with additional wires, but my RPi4 still reports under-voltage (red led blinks and then turns off) in one of the cars. Also it turns out the two of the USB ports are powered independently, so I was wrong with my conclusion that there was no voltage dip because I measured at the unused port. Measuring at the same port at RPi I get consistent 4.8V, which is already on the low side, with possible short dips that my DMM can't capture, but RPi's PMIC can

Anyway, I've ordered a $15 CarPlay+AA Wireless adapter based on Allwinner v851 from Aliexpress and it works well so far in both cars without audio dropouts or power issues

closed 2025-09-10 by kadimgh