aa-proxy/aa-proxy-rs #61: Possible future feature: Android auto to carplay gateway

spokosjr, 2025-09-20:
Hello, I own a Renault Megane with the R-Link 1 System. This only supports wired android auto as the display runs a really old version of android 2.2, I have recently changed my phone to an iPhone and I was wondering if there was a way that a translation layer could be built on the raspberry pi so that it can connect to the iPhone wirelessly (pretending to be a wireless carplay capable headunit) and then translate the protocol to wired android auto that my car's headunit can accept. Any help on what i could create, code wise to implement such a thing.

gamelaster, 2025-09-20:
This would be a heavy task to do + cause delays and several issues. So, by my opinion, not possible to do right now.

manio, 2025-09-20:
I fully agree with @gamelaster.
Additionally, I’m personally not very fond of the Apple ecosystem for various reasons, so I don’t intend to work on CarPlay integration in this project.
Closing.

closed 2025-09-20 by manio