aa-proxy/aa-proxy-rs #65: [SUGGESTION] WiCAN Pro (open-source OBD dongle) integration support via USB, Bluetooth, or WiFi?

Hedda, 2025-09-25:
Spotten in the readme for aa-proxy-rs that you are can connect wirelessly to some kind of Bluetooth OBD2 dongle (CAN-bus adapter), however I would like to suggest you specifically consider  looking into "WiCAN Pro" (open-source OBD dongle) integration support via USB, Bluetooth, or WiFi.

* https://www.crowdsupply.com/meatpi-electronics/wican-pro

WiCAN Pro is an OBD-II adapter that is a open-source hardware and open-source firmware project from meatPi (who is a single developer with the help of a small community of other volenteering developers and users). It is based on Espressif 's ESP32 and feature USB, Bluetooth, or WiFi.

* https://github.com/meatpiHQ/wican-fw/

  * https://github.com/meatpiHQ/wican-fw/issues

  “_WiCAN Pro - Advanced OBD scanner for next-generation vehicle diagnostics and Home Assistant integration. Built on the robust ESP32-S3 platform, WiCAN Pro runs the powerful, open-source WiCAN firmware. The inclusion of a USB-C connector for firmware flashing and a microSD card slot for data logging further enhances its functionality._”

  * https://meatpihq.github.io/wican-fw/

FYI, the developer also has an ongoing crowdfunding campaign for "ESPNetLink LTE" which is a 4G + GPS module for WiCAN Pro:

* https://www.crowdsupply.com/meatpi-electronics/espnetlink

  “_ESPNetLink, the LTE connectivity upgrade for your WiCAN Pro. With ESPNetLink, you can access your vehicle data securely and privately, directly from Home Assistant, without relying on third-party cloud services._”

  * https://www.crowdsupply.com/meatpi-electronics/espnetlink/updates/our-new-campaign-is-live-take-wican-pro-anywhere-with-espnetlink

WiCAN as an open-source product and project is growing a niche following amoung Home Assistant users and now has an "official" ingeration for:

* https://github.com/jay-oswald/ha-wican

  * https://community.home-assistant.io/t/wican-esp32-based-open-source-obd2-obd-ii-vehicle-car-or-truck-can-bus-diagnostics-to-wifi-ble-usb-adapter-for-home-assistant-integration/456098/

Hedda, 2025-09-25:
> Hehe, are you talking about this? :) https://github.com/aa-proxy/aa-proxy-wican

Ah, nice!

manio, 2025-09-25:
> Do you also also plan to port your canze-rs project to it as well?

It is ported and working (in branch) but still not cleaned up...
But yes, I plan to do it as well.

Join our discord for more info...